ARM-software/ML-KWS-for-MCU #2: How can i run these models on MCU？

fengzi0, 2017-12-20:
Thanks for your open source project. In your paper，you have deployed the KWS application on Cortex-M7. But there's only tensorflow models in this project, so what should i do to run these models on Cortex-M7？ Which system should the Cortex-M7 running, RTOS/Ubuntu？And how can i port tensorflow to Cortex-M7？ I will appreciate if you can give me some suggestions！

fabiovito, 2017-12-20:
I have read this paper and i think it's very interesting. I tried the python(tensorflow) code, but it's possible try also the c code for cortex?

navsuda, 2017-12-20:
Thank you for your interest in this work. We are planning to open source the code soon, along with examples of how to deploy them on one of the Mbed boards. 

In the mean time, for the Tensorflow model deployment on microcontrollers, Pete Warden from the Tensorflow team has a nice blog on it: https://petewarden.com/2017/05/08/running-tensorflow-graphs-on-microcontrollers/

closed 2018-01-03 by fengzi0

navsuda, 2018-01-25:
Hi @fengzi0 / @fabiovito ,
Examples are updated in the repo. Hope it helps.

ccnankai, 2018-11-23:
@fengzi0 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?