ARM-software/ML-KWS-for-MCU #3: No op named DecodeWav in defined operations.

Shahnawax, 2017-12-21:
Hi Naveen

First of all I like to thank you for uploading this nice set of example networks for keyword spotting. It is a great help and good starting point for many people like me.

I am a newbie to TensorFlow and while trying to load the models I get an error. Here is the code that I am using to load the model
```
import tensorflow as tf
from tensorflow.python.platform import gfile
with tf.Session() as sess:
    model_filename ='./Pretrained_models/DNN/DNN_S.pb'
    with gfile.FastGFile(model_filename, 'rb') as f:
        graph_def = tf.GraphDef()
        graph_def.ParseFromString(f.read())
        g_in = tf.import_graph_def(graph_def)
```
and here is the error
```
g_in = tf.import_graph_def(graph_def)
  File "/home/users/stageg4m/venv/pynn/lib/python3.5/site-packages/tensorflow/python/framework/importer.py", line 285, in import_graph_def
    raise ValueError('No op named %s in defined operations.' % node.op)
ValueError: No op named DecodeWav in defined operations.
```
While searching for the solution online I found out that one of the reasons for this error can be the usage of a different versions for freezing and loading the models. If this is the reason can you please tell me the version of tensorflow you used for creating the models? If not can you please help me to solve this problem? I will be grateful.

navsuda, 2017-12-21:
@Shahnawax , it seems you are using an older version of tensorflow. Please use TF 1.4 for loading the trained models.

Shahnawax, 2017-12-22:
Dear Naveen, thank you for quick reply.

I tried tensorflow version `1.0.1`, `1.4.0` and `1.4.1`. With all versions I am getting the same error. :(

navsuda, 2017-12-25:
Hi @Shahnawax, it may be related to this: https://github.com/tensorflow/tensorflow/issues/12657. Also, can you try generating the .pb file on the same machine as you are testing it. (Training shouldn't take very long, ~1hr on a decent GPU).

NanWang0024, 2018-01-26:
To confirm, tensorflow/tensorflow#12657 successfully solved this problem.

A side note: better to use TF 1.5 since another error (_No module named 'tensorflow.python.ops.gen_audio_ops'_) showed up in 1.4 as described in https://github.com/tensorflow/tensorflow/issues/14004.
Successfully loaded the models with a nightly build (tf_nightly-1.head-cp36-cp36m-win_amd64.whl).

navsuda, 2018-02-08:
Thanks @NanWang0024 . 
Closing the issue as it is fixed in TF 1.5.

closed 2018-02-08 by navsuda

ccnankai, 2018-11-23:
@Shahnawax 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?