ARM-software/ML-KWS-for-MCU #4: Question: Extend to larger network

donigian, 2018-01-02:
Thanks for extending the TF audio speech command app & making this available.

1) Say the hardware specs allow for a larger network to be trained (think XL or 2XL in comparison to S, M, L sizes mentioned in paper), what would be the best way to train such a network?

I tried the following:
`python train.py --model_architecture ds_cnn --model_size_info 9 276 10 4 2 1 276 3 3 2 2 276 3 3 1 1 276 3 3 1 1 276 3 3 1 1 276 3 3 1 1 276 3 3 1 1 276 3 3 1 1 276 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DS_CNN_9layer/retrain_logs --train_dir work/DS_CNN_9layer/training`

But increasing the size of the NN didn't result in improved accuracy on validation or test set. How would you explain this? Wouldn't a larger MOps yield better accuracy at cost of latency?

2) The training commands in `train_commands.txt` were super useful, wish more github projects would be as complete as this repo. However, after running a few of the training commands, I noticed that validation accuracy approached 99 to 100%. I wonder how many iterations did you stop at (which checkpoint file did you use) to generate the pre-trained models you provided?

navsuda, 2018-01-03:
Hi @donigian,
Glad that you find this repo useful. 
1) Your training command should work for the 9-layer network with 276 features in each layer. The number of features of 64, 172 and 276 come from the memory/compute bounds of the 3 classes of systems considered in this work. You may also try different features per layer, but that increases the hyperparameter search space. From our experiments, it seems that validation accuracy saturates around ~95% for the memory/compute bounds we considered and in order to get higher accuracy you may have to go very deep (for instance, ResNet-152?) or try new network topologies (DenseNet, NasNet,...). 
2) Did you mean "training" accuracy (and not "validation" accuracy) reaching ~100% after initial few training iterations? It might be the case that network is overfitting to training set and dropout may help reduce the overfitting. The train.py script saves the checkpoint with highest accuracy on validation set and the pre-trained model (.pb) files are generated from the last saved checkpoint.

donigian, 2018-01-10:
When you execute `python train.py --model_architecture ds_cnn --model_size_info 6 276 10 4 2 1 276 3 3 2 2 276 3 3 1 1 276 3 3 1 1 276 3 3 1 1 276 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DS_CNN/DS_CNN3/retrain_logs --train_dir work/DS_CNN/DS_CNN3/training `...

what iteration does the command end at? What's the name of the generated checkpoint file?

bukosabino, 2018-01-10:
It will take 30000 iterations.

The name of checkpoint file will be different for each run. Example: ds_cnn_xxxx.ckpt-yyyyy

where xxxx = max score
where yyyyy = iteration of max score

donigian, 2018-01-10:
```INFO:tensorflow:So far the best validation accuracy is 94.02%
INFO:tensorflow:set_size=8536
INFO:tensorflow:Confusion Matrix:
INFO:tensorflow:Final test accuracy = 94.47% (N=8536)```

closed 2018-01-10 by donigian

ccnankai, 2018-11-23:
@donigian 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?