ARM-software/ML-KWS-for-MCU #5: Log-mel filter bank features

bukosabino, 2018-01-03:
Hi! Thank you for share your ideas and code. I'm newbie in DL and I'm learning a lot with your material.

I would like run this examples with log-mel filterbanks as features to the neural network input.

How I can do it?

navsuda, 2018-01-03:
Hi @bukosabino 

Glad that you find it useful.

1) Please see https://www.tensorflow.org/api_guides/python/contrib.signal for an example on how to use tf.contrib.signal to compute log-mel spectrograms. You will have to replace the mfcc call (https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/input_data.py#L382) with this log-mel spectrogram computation. 
2) The other possible way would be to use tf.py_func to wrap logfbank function from https://github.com/jameslyons/python_speech_features repo.

bukosabino, 2018-01-04:
Yeah. It works fine.

Thank you @navsuda

closed 2018-01-04 by bukosabino