ARM-software/ML-KWS-for-MCU #7: Quantization

pribadihcr, 2018-01-15:
Hi,

In conclusion of the paper, it states "We quantized representative trained 32-bit floating-point KWS models into 8-bit fixed-point versions... ".

Is the pretrained model already in 8 bit fixed point ??

navsuda, 2018-01-16:
Hi @pribadihcr 
The pretrained models are 32-bit floating-point models.

pribadihcr, 2018-01-16:
Hi @navsuda,

Is there tutorial to convert it to 8 bit?

coolo, 2018-01-16:
I found https://github.com/tensorflow/tensorflow/tree/master/tensorflow/tools/graph_transforms/#common-use-cases an interesting read, but I agree that the paper reads as if there was done more magic than that :)

navsuda, 2018-01-18:
Hi @pribadihcr,
There is no tutorial at this point, but the basic idea of quantization we use used is discussed in this paper: [Going deeper with embedded FPGA platform for convolutional neural network](https://pdfs.semanticscholar.org/c382/406fd8db2744b2a609837395e5da05e1d2ed.pdf) (section 5.1). I will put a guide on how to quantize the models when we release the end-to-end example running on microcontrollers.
@coolo, Its no magic :) but just rounding off weights and activations (or layer outputs) to the nearest power of 2 using `tf.fake_quant_with_min_max_args` op.

navsuda, 2018-01-25:
Hi @pribadihcr, 
Updated some example code for Cortex-M and a quick guide on quantization. Hope it helps.

coolo, 2018-01-25:
Very interesting stuff. Thanks!

pribadihcr, 2018-01-25:
@navsuda 

great. Thanks!

closed 2018-03-06 by navsuda

ccnankai, 2018-11-23:
Hi,@coolo  
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
 how to  the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0...   ) ?