ARM-software/ML-KWS-for-MCU #9: Training the model with my own data

zhaoforever, 2018-02-08:
Hi,
    As the train.py describes, the program support training with my own data. However, when I use my own .wav files by --data_dir, it says IOError: CRC check failed 0x7668da3b != 0x316d6d75L. Have you ever test the program with some other database instead of the google recordings? Thanks.

navsuda, 2018-02-08:
Hi @zhaoforever,
It works fine on a different dataset. To train on your own data, use "--data_url= " (i.e. data_url=<space>) along with --data_dir as given in comments [here](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/train.py#L102). Otherwise, the script tries to download the speech commands dataset into your data_dir (and cause CRC check fail error, if interrupted before the full download of the speech commands dataset).

zhaoforever, 2018-02-09:
@navsuda Thanks for your advice and the program runs well using both the google dataset and my own. But when I use the "python train.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000" command to get the .pb file of google database, the recognition results seems terrible comparing with the pretrained models. Does anything else need to be done?

navsuda, 2018-02-12:
Hi @zhaoforever,
What accuracy are you getting? This model should give close to 94% accuracy with those parameters.

zhaoforever, 2018-02-23:
@navsuda The accuracy of the training result about 94%. But when I run the label_wav.py using the trained  model, the accuracy is below 50%.

zhaoforever, 2018-02-23:
@navsuda Thanks for your help. The accuracy is low as the window size is not set correctly.

sailor88128, 2018-03-02:
Hi @zhaoforever ,
My own data are used to train the model, the accuracy 95.19. Then, the following command to generate pb model file used. Then the following error appears, what may be the reason?

```
python train.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DNN/DNN1/retrain_logs --wanted_words 'help,stop,start,goup,others'  --train_dir ./data/ --data_url= --data_dir ./data

python freeze.py --model_architecture dnn --model_size_info 144 144 144 --checkpoint=./data/best/dnn_9519.ckpt-9600 --output_file dnn_kws.pb

InvalidArgumentError (see above for traceback): Assign requires shapes of both tensors to match. lhs shape= [3920,144] rhs shape= [250,144]
```

zhaoforever, 2018-03-02:
@sailor88128 When using file freeze.py , you also need the following parameters  "--dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40" .

sailor88128, 2018-03-05:
@zhaoforever  It's OK now. thanks!

navsuda, 2018-03-06:
Thanks @zhaoforever. 
Closing this issue

upcoders, 2020-07-19:
In my case the error was due to difference between number of `wanted_words` with default words, so I resolved that with adding `--wanted_words` switch for `freeze.py` and the error cleared out.

> Hi @zhaoforever ,
> My own data are used to train the model, the accuracy 95.19. Then, the following command to generate pb model file used. Then the following error appears, what may be the reason? [...]

In my case the error was due to difference between number of `wanted_words` with default words which is 12, so I resolved that with adding `--wanted_words` switch for `freeze.py` and the error cleared out.