ARM-software/ML-KWS-for-MCU #10: How to use the LFBEs in tensorflow envoirnment

Shahnawax, 2018-02-28:
Dear Naveen

I was able to run this code and train the networks on a different set of classes. However, in the paper, Hello edge it was mentioned that we can train either on LFBE or on MFCCs. Can you please guide me about the preprocessing using logmel filter bank energies (LFBEs)? I am new to tensorflow and having a hard time to understand the gen_audio_ops.
I tried to use librosa but I am not getting meaningful results. Any kind of help will be highly appreciated. Thank you.

navsuda, 2018-03-06:
Hi @Shahnawax, please see [this](https://github.com/ARM-software/ML-KWS-for-MCU/issues/5#issuecomment-355060532)

Shahnawax, 2018-03-06:
Hi @navsuda  Thank you for your help. I did the same on Thursday and it works like a charm 👍

navsuda, 2018-03-06:
@Shahnawax ,
Glad it worked. 
Closing the issue.

closed 2018-03-06 by navsuda