ARM-software/ML-KWS-for-MCU #11: Is the model used only for the trainers

sunjunlishi, 2018-03-06:
Is the model used only for the trainers, is the input voice required, 16000HZ, 16bit, wav

navsuda, 2018-03-06:
@sunjunlishi,
Did you mean training? The training (or inference) should work fine with other sampling frequencies also as long as all training samples are consistent with same sampling frequency. I haven't tried if it supports other than 16-bit wav files, but [this](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/input_data.py#L352) and [this](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/core/kernels/decode_wav_op.cc#L53) suggest that the it is limited to 16-bit wav files.

sunjunlishi, 2018-03-07:
ok，it is good。it  must  16000HZ,16bit.
another question?
predictions, = sess.run(softmax_tensor, {input_layer_name: wav_data})
'wav_data' , Is its binary voice data as an input to the network

navsuda, 2018-03-07:
Correct. Please see [here](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/label_wav.py#L97)

sunjunlishi, 2018-03-07:
Placeholder DecodeWav AudioSpectrogram Mfcc Const Reshape Const Reshape Const Const Conv2D BiasAdd Const Const Const Const FusedBatchNorm Relu Const DepthwiseConv2dNative Const BiasAdd Const Const Const Const FusedBatchNorm Relu Const Const Conv2D BiasAdd Const Const Const Const FusedBatchNorm Relu Const DepthwiseConv2dNative Const BiasAdd Const Const Const Const FusedBatchNorm Relu Const Const Conv2D BiasAdd Const Const Const Const FusedBatchNorm Relu Const DepthwiseConv2dNative Const BiasAdd Const Const Const Const FusedBatchNorm Relu Const Const Conv2D BiasAdd Const Const Const Const FusedBatchNorm Relu Const DepthwiseConv2dNative Const BiasAdd Const Const Const Const FusedBatchNorm Relu Const Const Conv2D BiasAdd Const Const Const Const FusedBatchNorm Relu Const DepthwiseConv2dNative Const BiasAdd Const Const Const Const FusedBatchNorm Relu Const Const Conv2D BiasAdd Const Const Const Const FusedBatchNorm Relu AvgPool Squeeze Const Const MatMul BiasAdd Softmax

sunjunlishi, 2018-03-07:
Can these network layers be implemented in pure c++? Tensorflow code is too huge, and ARM's library can't cross platform. It's a bit depressed. But Your training template is excellent.