ARM-software/ML-KWS-for-MCU #14: Stack overflow

wangshuai1320, 2018-03-16:
The "int16_t audio_buffer[16000]=WAVE_DATA;" should not be defined in the main function because it will result in stack overflow when running with some other boards such K64F.
This variable should defined as global variables and store in heap.

navsuda, 2018-03-19:
Hi @Hust-ShuaiWang,
It worked fine on K64F board using mbed-cli and GCC compiler. What compiler/tool-chain did you use?
I agree, some other boards may face this issue if the stack space is small. In such cases, it is recommended to store in heap by allocating the memory in main function.

Hust-ShuaiWang, 2018-03-20:
Thank you for your reply and I want to ask you some questions which puzzle me.
Could you tell me the  meaning and effect of parameters of bias_shift and out_shift in arm_fully_connected_q7(...,  const uint16_t bias_shift,  const uint16_t out_shift, ........)?(No ARM_MATH_DSP)

Thank you very much.

The format of bias ( ....const q7_t *bias..... )is q7_t and you convert it to q31_t ( int ip_out = ((q31_t)(bias[i]) << bias_shift) + NN_ROUND(out_shift);).Inorder to prevent errors, we have to move (q31_t)bias[i] 24 bits left,  but you move it by parameters as bias_shift and I can not understand the meaning of this parameter.
Maybe I have did some wrong understanding of the parameter and this parameter has some other functions.
Could explain me the role of these two parameters?
Thank you very much and I wish to get your apply.😊

navsuda, 2018-03-27:
These two parameters are used to adjust (shift) the weights and data ranges so that they can be represented in 8-bit fixed-point representation without much loss in accuracy.
Please see this [discussion](https://github.com/ARM-software/CMSIS_5/issues/327#issuecomment-369433866).

The basic idea is as follows:
Consider this equation, which is common for fully-connected layers or convolution layers: 
Layer_output = [sum(layer_input x weights) + (bias<<bias_left_shift)] >> out_right_shift. 
The key is to make sure that values that we are summing up have same [fixed-point representation](https://en.wikipedia.org/wiki/Q_(number_format)) (i.e. same number of bits for fractional part). 
For example, the arm_fully_connected_q7 function call [here](https://github.com/ARM-software/ML-KWS-for-MCU/blob/8ea22926f743f53c7d17d9c73eb2f1b22257ebe2/Deployment/Source/DNN/dnn.cpp#L67), from the quantization sweeps we get the input is of Q5.2 in fixed-point format or 2 bits after decimal point (i.e. ranges from -32,31.75) and the expected output format for maximum accuracy is Q5.2 (i.e. ranges from -32,32). Using quant_test.py, we get the quantized weights are of the format Q0.7 (i.e. in range -1,1) and the quantized biases have 8 bits for fractional point (i.e. range -0.5,0.5). So,  the product (layer_input x weights) will have 9 bits in the fractional part (Q5.2 x Q0.7 = Qx.9) and biases need to be shifted left by 1 to get to the same representation (i.e. Qx.8<<1 = Qy.9). The expected layer output has 2 bits for fractional part (Q5.2), so the product (layer_input x weights) needs to be shifted right by 7 to get to Q5.2 format. 
Hope it helps (a bit 😊).

Hust-ShuaiWang, 2018-04-23:
Thank you very much for your reply.

The input format of first layer is Q5.2 (which is gotten from mfcc),the weights of first layer is Q0.7 ,so the product of (input * weights) is Qx.9.The bias is Q0.8,the expected format of output is Q5.2.
So the bias_shift is 1,out_shift is 7 and the format of first layer's output is Q5.2 which is the input of second layer(ReLU has no effect on data format ).The main purpose of the shift operation is to make 2 parameters in same format before operation( + - * /).

**The second layer has the same format of input data as first layer, but the bias_shift and out_shift is 2 and 8**.In my understanding, weights and bias in different layers has different ranges and the second layer's weights and bias has different formats as first layer. So the bias_shift and out_shift is different in different layers.

If I have done right understanding in that and your weights and bias in different layers are surely different, would you like to mark formats of weights and formats in dnn_weights.h ? I think that will help people make understanding more easily.
Maybe I have did some wrong understanding in the 2 parameters again and please forgive my slowness haha😄~  Would you like to tell me why those 2 parameters are different in layer 1 and 2?
Thank  you very much~

navsuda, 2018-04-26:
Hi @Hust-ShuaiWang,
You got it all right. The shift parameters may be different in different layers (even if they have same input output ranges) because of the different ranges of bias/weights. (Too many differents in the same sentence 😄). The output of quant_test.py shows the ranges of bias/weights.

navsuda, 2018-07-11:
Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda

ccnankai, 2018-11-23:
@Hust-ShuaiWang 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?

saichand07, 2019-04-09:
@Hust-ShuaiWang 
Thanks., you explained your understanding very nicely, 
Could you please tell me how did you find out the input layer format? From where we can get to know the formats of weights, bias?
Can you tell me am I right or wrong about this formats?

**fc1_W_0 number of wts/bias: (3920, 128) dec bits: 6 max: (1.875,1.87343) min: (-1.53125,-1.53759)

Q1.3 is format of weights of first layer (I have only considered max from (1.875,1.87343)), how to make use of min (-1.53125,-1.53759)**

**fc1_b_0 number of wts/bias: (128,) dec bits: 6 max: (0.71875,0.723896) min: (-1.09375,-1.09377)

Q1.6 is the bias format (only considered the max(0.723896))**