ARM-software/ML-KWS-for-MCU #16: Issue with quant_test.py, and request for DS_CNN

daisukelab, 2018-03-27:
Hi, I'm trying to get my DNN work on Disco board.

1. I have problem with quant_test.py, it came across this error.
Is it possible to show what could avoid this please?

```sh
$ python quant_test.py --data_url= --data_dir=/dataset/speech_commands_v0.01/ --model_architecture dnn --model_size_info 128 128 128 --checkpoint /tmp/speech_commands_train/best/dnn_8480.ckpt-17200
Traceback (most recent call last):
  File "quant_test.py", line 305, in <module>
    tf.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "/Users/foo/anaconda3/lib/python3.6/site-packages/tensorflow/python/platform/app.py", line 124, in run
    _sys.exit(main(argv))
  File "quant_test.py", line 203, in main
    FLAGS.model_architecture, FLAGS.model_size_info)
  File "quant_test.py", line 79, in run_quant_inference
    is_training=False)
  File "/Users/foo/lab/ML-KWS-for-MCU/quant_models.py", line 115, in create_model
    act_max, is_training)
  File "/Users/foo/lab/ML-KWS-for-MCU/quant_models.py", line 162, in create_dnn_model
    if(act_max[i]!=0):
IndexError: list index out of range
```

Before trying this, I have trained my network as follows. It was successful.

```sh
python train.py --data_url= --data_dir=/dataset/speech_commands_v0.01 --model_architecture dnn --model_size_info 128 128 128
```

2. I understand that only DNN is supported by this for now, we are very very happy if you could open quant_test.py as well as Deployment codes. Thank you!

navsuda, 2018-03-27:
Hi @daisukelab,
1) The quant_test.py helps you quantize the weights and layer input/outputs or activation data to Fixed-point representation, but you have to provide the quantization ranges for the activation data using "--act_max" argument as mentioned at the end of the [quantization guide](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Quant_guide.md). quant_test.py then runs the model by quantizing the data using the ranges you provided and gives you the accuracy. You have to find the optimal parameters for the "--act_max" that minimizes the accuracy loss because of quantization.
2) DS-CNN model will be added in the coming weeks.

daisukelab, 2018-03-28:
Hi @navsuda, thank you for your feedback.
1. It worked as follows.
2. We are very happy to hear about that, I cannot wait to work with it!
Thanks again!

```sh
$ python quant_test.py --data_url= --data_dir=/dataset/speech_commands_v0.01/ --model_architecture dnn --model_size_info 128 128 128 --checkpoint /tmp/speech_commands_train/best/dnn_8457.ckpt-18000 --act_max 32 0 0 0 0
/home/foo/.pyenv/versions/anaconda3-5.0.1/envs/a3ml/lib/python3.6/importlib/_bootstrap.py:219: RuntimeWarning: compiletime version 3.5 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.6
  return f(*args, **kwds)
/home/foo/.pyenv/versions/anaconda3-5.0.1/envs/a3ml/lib/python3.6/site-packages/h5py/__init__.py:34: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
2018-03-28 11:24:18.190228: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: SSE4.1 SSE4.2 AVX AVX2 AVX512F FMA
2018-03-28 11:24:19.943098: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1030] Found device 0 with properties: 
name: GeForce GTX 1080 Ti major: 6 minor: 1 memoryClockRate(GHz): 1.6325
pciBusID: 0000:17:00.0
totalMemory: 10.91GiB freeMemory: 10.75GiB
2018-03-28 11:24:20.250737: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1030] Found device 1 with properties: 
name: GeForce GTX 1080 Ti major: 6 minor: 1 memoryClockRate(GHz): 1.645
pciBusID: 0000:65:00.0
totalMemory: 10.91GiB freeMemory: 10.75GiB
2018-03-28 11:24:20.252188: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1045] Device peer to peer matrix
2018-03-28 11:24:20.252322: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1051] DMA: 0 1 
2018-03-28 11:24:20.252342: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1061] 0:   Y Y 
2018-03-28 11:24:20.252369: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1061] 1:   Y Y 
2018-03-28 11:24:20.252441: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1120] Creating TensorFlow device (/device:GPU:0) -> (device: 0, name: GeForce GTX 1080 Ti, pci bus id: 0000:17:00.0, compute capability: 6.1)
2018-03-28 11:24:20.252455: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1120] Creating TensorFlow device (/device:GPU:1) -> (device: 1, name: GeForce GTX 1080 Ti, pci bus id: 0000:65:00.0, compute capability: 6.1)
2018-03-28 11:24:20.874305: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1120] Creating TensorFlow device (/device:GPU:0) -> (device: 0, name: GeForce GTX 1080 Ti, pci bus id: 0000:17:00.0, compute capability: 6.1)
2018-03-28 11:24:20.874353: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1120] Creating TensorFlow device (/device:GPU:1) -> (device: 1, name: GeForce GTX 1080 Ti, pci bus id: 0000:65:00.0, compute capability: 6.1)
INFO:tensorflow:Restoring parameters from /tmp/speech_commands_train/best/dnn_8457.ckpt-18000
fc1_W_0 number of wts/bias: (3920, 128) dec bits: 6 max: (1.109375,1.1144608) min: (-0.921875,-0.92070407)
fc1_b_0 number of wts/bias: (128,) dec bits: 7 max: (0.8359375,0.83384776) min: (-0.8671875,-0.8654579)
fc2_W_0 number of wts/bias: (128, 128) dec bits: 7 max: (0.4765625,0.47322175) min: (-0.53125,-0.53390217)
fc2_b_0 number of wts/bias: (128,) dec bits: 6 max: (0.9375,0.9310091) min: (-1.046875,-1.0456378)
fc3_W_0 number of wts/bias: (128, 128) dec bits: 8 max: (0.41015625,0.40897876) min: (-0.48828125,-0.48746592)
fc3_b_0 number of wts/bias: (128,) dec bits: 7 max: (0.953125,0.95504075) min: (-0.3203125,-0.31765613)
final_fc_0 number of wts/bias: (128, 12) dec bits: 7 max: (0.2421875,0.24324849) min: (-0.578125,-0.5809378)
Variable_0 number of wts/bias: (12,) dec bits: 7 max: (0.65625,0.6553219) min: (-0.3125,-0.31391922)
INFO:tensorflow:set_size=22246
INFO:tensorflow:Confusion Matrix:
 [[1831    0    0    0    0    0    0    0    0    0    0    0]
 [   5 1735   11   18    5   22    9   20   16    2   14   11]
 [   2   12 1760    9    6    2   16    0    4    0    1    0]
 [   9   16    5 1735    6    8    1    1    3    0    3   61]
 [  11   11    5    0 1706    0    1    4    2   18   55   12]
 [   8    5    2   20    3 1779    2    0    2    1   18    9]
 [   8   11   20    4    6    2 1768   11    1    7    3    6]
 [   7   11    0    2    1    3    5 1773    0    4    1    7]
 [   6   24    2    1   10    5    0    4 1805   16    6    5]
 [  16    9    0    0   37    2    0    2    9 1795   14    2]
 [  10    5    0    3   13   11    5    0    4    2 1859   12]
 [   2   14    4   86    4   23    1    2    1    1   11 1709]]
INFO:tensorflow:Training accuracy = 95.55% (N=22246)
INFO:tensorflow:set_size=3093
INFO:tensorflow:Confusion Matrix:
 [[258   0   0   0   0   0   0   0   0   0   0   0]
 [  1 186   5   8   3   7   3  13  11   6   6   9]
 [  1   7 233   6   0   1   6   2   1   0   0   4]
 [  0  15   1 219   1   6   2   1   1   0   2  22]
 [  0   3   0   1 222   0   4   0   3   9  12   6]
 [  0   6   3  20   0 217   1   0   2   0   3  12]
 [  0   5  24   2   2   0 202   9   0   2   1   0]
 [  0   8   1   0   2   0   3 239   1   1   0   1]
 [  1  14   0   1   1   3   0   1 221   9   1   5]
 [  0   3   1   1  16   0   1   1   3 225   2   3]
 [  1   6   0   0   9   0   0   1   1   4 222   2]
 [  2   9   0  33   6   8   0   1   2   1   0 198]]
[... cut ...]
 [  0  10   4 187   1  21   1   0   0   0   0  28]
 [  0   7   1   3 223   1   6   0   0   8  16   7]
 [  0  12   1  18   3 201   1   1   0   0   2  14]
 [  0  10  13   4   4   3 227   5   0   1   0   0]
 [  1  22   1   0   2   0   1 226   0   2   0   4]
 [  0   8   0   0   3   2   0   1 222   9   0   1]
 [  0   9   0   0  14   1   2   0  10 221   2   3]
 [  0   3   1   4   4   3   0   1   1   4 221   7]
 [  0   8   1  32   5   9   5   1   1   2   1 186]]
INFO:tensorflow:Test accuracy = 84.23% (N=3081)
```

navsuda, 2018-04-25:
Hi @daisukelab,
Added DS_CNN model to the repo. Hope it helps.

closed 2018-04-25 by navsuda

daisukelab, 2018-04-26:
Hi @navsuda,
Thank you very much. I really appreciate.