ARM-software/ML-KWS-for-MCU #17: Kernel size of DS-CNN's Conv1

blmd-niz, 2018-04-07:
Hi, I read throughout the paper but it's still not clear about kernel size, so let me ask:

1. How did you decide your kernel size of DS-CNN (4 x 10)?
2.Is it "time by feature" or "feature by time"? It looks like size for feature is 4, and size for time is 10 in your code. But for me, it is more reasonable if 10 is for feature and 4 is for time.

This is because it seems not to be clear what is the best kernel representation especially for speech, questioned by this article:
https://towardsdatascience.com/whats-wrong-with-spectrograms-and-cnns-for-audio-processing-311377d7ccd

I appreciate if you could share any of your thought when you were elaborating your fine DS-CNN.
Thank you.

navsuda, 2018-04-16:
Hi @blmd-niz,
Thanks for raising some fundamental questions on spatial-variance property in frequency domain in speech features vs. spatial-invariance property in images. 
1) The filter dimensions are inspired from [this ](https://static.googleusercontent.com/media/research.google.com/en//pubs/archive/43969.pdf), [this](https://arxiv.org/ftp/arxiv/papers/1703/1703.05390.pdf) and [this](https://arxiv.org/pdf/1512.02595.pdf) papers, combined with some hyperparameter search to find the best dimensions that gives best accuracy for small number of parameters and Ops. 
2) 10 dimension is in time and 4 is in frequency axis. In my opinion, it works out fine because convolutional layer removes speaker variability by consolidating the useful features in the initial layers, followed by fully-connected layers or Recurrent layers to perform the final classification.

closed 2018-04-19 by blmd-niz

xxoospring, 2019-05-08:
@blmd-niz , could you tell me where do you get the details of the structure of this dscnn model? Thanks

blmd-niz, 2019-05-08:
@xxoospring Hi I don't remember correctly but it would be by reading source code. For example:
https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/models.py

saichand07, 2019-05-08:
Hi @blmd-niz 
have you deployed ds_cnn model (trained by you) on board, does it working properly ? 

Thanks

blmd-niz, 2019-05-08:
@saichand07 No I haven't tried yet... unfortunately my log shows only normal DNN was tried...

saichand07, 2019-05-08:
@Thanks for the quick reply

Do you have any idea how to deal with bias negative shifts, and how to deal with the situation like shift values larger than 8 bits ?

blmd-niz, 2019-05-09:
I'm sorry but I'm not familiar with this repo recently, so I don't have idea. I hope you would find soon...