ARM-software/ML-KWS-for-MCU #18: GCC compilation

kdpatino, 2018-04-10:
Hello,

* Could I compile this project with GCC? 
* Do you have any examples of **CMAKE** or any open source tool?

Thank You in advance!

Kevin Patino

navsuda, 2018-04-16:
Hi @kdpatino,
- It is possible to compile the project (specific to a particular dev board) using gcc-arm tools. This link has details on how to export an mbed-online project to compile locally using gcc tools: https://os.mbed.com/handbook/Exporting-to-GCC-ARM-Embedded
- Unfortunately, we do not have examples of CMAKE to compile the examples.

oneway3124, 2018-04-27:
Hi navsuda,

I have one same development as this site referred, but the mbed compile is so complex that I have cost one day for this problem.

david.wang

navsuda, 2018-05-22:
Added an example to compile without the need for mbed-cli here: [https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment/Examples/simple_test_k64f_gcc](https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment/Examples/simple_test_k64f_gcc). Hope it helps.

oneway3124, 2018-05-23:
@navsuda 
Your example help me compile successfully.
What's more, I want to know the file folder which you added . Where did they come from ?

navsuda, 2018-06-22:
Hi @oneway3124,
You can get that by creating a simple hello world project on [mbed online compiler](https://os.mbed.com/compiler/) with the specific board you are targeting and then export with Make-GNU-ARM toolchain as described [here](https://os.mbed.com/handbook/Exporting-to-Make).

navsuda, 2018-07-11:
Closing the issue, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda