ARM-software/ML-KWS-for-MCU #19: How to execute training and other steps to generate the example

cledic, 2018-04-15:
Hello to all,
I'm trying to execute again the training and the other steps to reproduce the .pb and weights.h files.
I use this commands:

python train.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DNN/DNN1/retrain_logs --train_dir work/DNN/DNN1/training 

python test.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DNN/DNN1/retrain_logs --train_dir work/DNN/DNN1/training --checkpoint /home/embedded/ML-KWS-for-MCU/work/DNN/DNN1/training/best/dnn_8448.ckpt-29600

python freeze.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DNN/DNN1/retrain_logs --train_dir work/DNN/DNN1/training --checkpoint /home/embedded/ML-KWS-for-MCU/work/DNN/DNN1/training/best/dnn_8448.ckpt-29600 --output_file dnn.pb

But running this last script I got an error:
python quant_test.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --checkpoint /home/embedded/ML-KWS-for-MCU/work/DNN/DNN1/training/best/dnn_8448.ckpt-29600

The error is:
Traceback (most recent call last):
  File "quant_test.py", line 305, in <module>
    tf.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "/home/embedded/.local/lib/python2.7/site-packages/tensorflow/python/platform/app.py", line 124, in run
    _sys.exit(main(argv))
  File "quant_test.py", line 203, in main
    FLAGS.model_architecture, FLAGS.model_size_info)
  File "quant_test.py", line 79, in run_quant_inference
    is_training=False)
  File "/home/embedded/ML-KWS-for-MCU/quant_models.py", line 115, in create_model
    act_max, is_training)
  File "/home/embedded/ML-KWS-for-MCU/quant_models.py", line 160, in create_dnn_model
    if(act_max[i]!=0):
IndexError: list index out of range

Please: where am I wrong? 
Can someone publish the correct commands? I want to rebuild and test the project and then I want to replace the sound with different sound, in my case urban sound, and repeat the build.

Thanks a lot.
Clemente

daisukelab, 2018-04-15:
Hi, this looks like the same with my issue resolved the other day.
https://github.com/ARM-software/ML-KWS-for-MCU/issues/16

Just adding ` --act_max 32 0 0 0 0` would solve your problem like this, please try:

```Python
python quant_test.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --checkpoint /home/embedded/ML-KWS-for-MCU/work/DNN/DNN1/training/best/dnn_8448.ckpt-29600 --act_max 32 0 0 0 0
```

cledic, 2018-04-15:
Hello Daisuke, thanks,
you are right, the quant_test.py now run without problem, but the resulting weights.h file is different from the original. The define are not the same. The new generated is like: "fc1_W_0" the original is: "IP1_BIAS"
Do you know why?

daisukelab, 2018-04-16:
Not sure why, but I think we can simply change the code to match the generated. Here's my change:

```C++
const q7_t DNN::ip1_wt[IP1_WT_DIM]=fc1_W_0;
const q7_t DNN::ip1_bias[IP1_OUT_DIM]=fc1_b_0;
const q7_t DNN::ip2_wt[IP2_WT_DIM]=fc2_W_0;
const q7_t DNN::ip2_bias[IP2_OUT_DIM]=fc2_b_0;
const q7_t DNN::ip3_wt[IP3_WT_DIM]=fc3_W_0;
const q7_t DNN::ip3_bias[IP3_OUT_DIM]=fc3_b_0;
const q7_t DNN::ip4_wt[IP4_WT_DIM]=final_fc_0;
const q7_t DNN::ip4_bias[OUT_DIM]=Variable_0;
```

Thanks to @navsuda, they are working in progress to add DS-CNN.
https://github.com/ARM-software/ML-KWS-for-MCU/issues/16#issuecomment-376586524

cledic, 2018-04-16:
Many thanks Daisuke, I will try soon!

cledic, 2018-04-17:
Hello Daisuke,
it works !!!
I put some doc here [1]. May be can help other people.

Thanks
Clemente
[1] https://github.com/cledic/ML-KWS-for-MCU/blob/master/README2.md

navsuda, 2018-04-26:
Thank you @daisukelab and @cledic.
Closing the issue.

closed 2018-04-26 by navsuda

ccnankai, 2018-11-23:
Hi @cledic  @daisukelab 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?