ARM-software/ML-KWS-for-MCU #20: Issue running with F411RE

mridup, 2018-04-16:
Hi,
I tried to execute the kws_simple_test example as given in https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment

Board used is NUCLEO_F411RE.
The program crashes in the following line:
KWS kws(audio_buffer,scratch_buffer);
in main.cpp of Examples/simple_test

Did not investigate further but these examples seem tested on Nucleo boards?

regards,
Mridu

mridup, 2018-04-19:
issue resolved when using global buffers.

closed 2018-04-19 by mridup

ccnankai, 2018-11-23:
@mridup 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?