ARM-software/ML-KWS-for-MCU #21: how to generate new input wav_data.h

fabiovito, 2018-04-19:
Hi,
how can i generate a new input audio example "wav_data.h"?
thanks.

navsuda, 2018-04-25:
Hi @fabiovito,
wav_data.h consists of raw audio data in 16-bit format. You can read the .wav file in python and write to .h file. The following code should work fine. 

import wave
audio = wave.open("your wav file")
signal = audio.readframes(-1)
signal = np.fromstring(signal, 'Int16')
with open("wav_data.h",'ab') as f:
  np.savetxt(f,signal,fmt="%d",delimiter=",",newline=",")

gamnes, 2018-05-31:
Hi @navsuda, would you be able to provide some more wav_data.h files to use for testing with the simple_test example? And what should the output of the default wav_data.h file be? Thanks:)

navsuda, 2018-06-22:
Hi @gamnes,
The output with default wav_data.h is "right". With the provided python script, you should be able to generate any number of examples by providing any audio file from the speech commands dataset.

navsuda, 2018-07-11:
Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda