ARM-software/ML-KWS-for-MCU #22: Incompatibility problem wiht mbed-os

Gurumin, 2018-04-20:
Hi:

I found that the case was not compatiolbe with mbed-os, if I use class mbed::timer(Timer T;), the code would cause a BusFault.
I have tested the case on borad FRDM-K64F and STM32L476-DISCO, whit mbed-os-5.5.3 and mbed-os-5.7.7, the sesults were same.
I noticed that even in the latest mbed-os, the lib cmsis_dsp was located in features/unsupported/dsp/cmsis_dsp.
Have you ever met the same problem?

Thanks!

navsuda, 2018-05-16:
Hi @Gurumin,
I haven't faced this issue with K64F using latest mbed-os. Can you update your mbed-cli and try on the latest mbed-os ?

navsuda, 2018-07-11:
Hi @Gurumin,
Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda

ccnankai, 2018-11-23:
@Gurumin 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?