ARM-software/ML-KWS-for-MCU #24: there are  one error when I try to deploy on target.

oneway3124, 2018-04-27:
1.Build and run a simple KWS inference
2.mbed new kws_simple_test --mbedlib
3.  ERROR  An error occurred while unpacking library archive ".bld\.rev-5aab5a7997ee.zip" in "c:\Users\Administrator\Desktop\ML-KWS-for-MCU\Deployment\kws_simple_test\mbed"

oneway3124, 2018-04-27:
This issue has be fixed. At the same time, when I was compiling the kws_realtime_test，
[image]
collect2.exe: error: ld returned 1 exit status

oneway3124, 2018-04-28:
Some times there is one error when I am executing the command "mbed deploy".
[mbed] ERROR: An error occurred while unpacking library archive ".bld\.rev-7130f322cb7e.zip" in xxx

zhuanaa, 2018-05-03:
did you come across "Couldn't find build tools in your program." at the step of "mbed new kws_simple_test --mbedlib"? Maybe this result in your problem.

navsuda, 2018-05-07:
Hi  @oneway3124 
It looks like it is related to [#516](https://github.com/ARMmbed/mbed-cli/issues/516), [#448](https://github.com/ARMmbed/mbed-cli/issues/448).

oneway3124, 2018-05-14:
@navsuda ,

Some check after your reviews, I think that these answers #516, #448, may not be the right ones. 
Can you give me some more information about mbed-cli?

navsuda, 2018-06-22:
Hi @oneway3124,
If you have solved your environment issue, please consider closing the issue.

navsuda, 2018-07-11:
Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda

jimmy9389, 2019-06-14:
Hi @oneway3124 I'm facing the exactly same problem you describe at first, could you please tell how you sovled it?
[image]