ARM-software/ML-KWS-for-MCU #25: Model size about DS_CNN_L.pb

WeldonWangwang, 2018-04-27:
hi, 
I just use DS-CNN model to test, but I can't ensure the value of "model_size_info".

So I want to read the variable value from DS_CNN_L.pb by:  
```
with tf.Graph().as_default():
    output_graph_def = tf.GraphDef()
    output_graph_path = './Pretrained_models/DS_CNN/DS_CNN_L.pb'
    #sess.graph.add_to_collection("input", mnist.test.images)

    with open(output_graph_path, "rb") as f:
        output_graph_def.ParseFromString(f.read())
        _ = tf.import_graph_def(output_graph_def, name="")

    with tf.Session() as sess:

        tf.initialize_all_variables().run()
        input_x = sess.graph.get_tensor_by_name("DS-CNN/fc1/weights:0")
        print input_x
        output = sess.graph.get_tensor_by_name("output:0")
        print output
```
  
But it returns error:
`KeyError: "The name 'DS-CNN/conv_1/weights:0' refers to a Tensor which does not exist. The operation, 'DS-CNN/conv_1/weights', does not exist in the graph."`

So I want to ask you what's  the model_size_info in Pretrained_models/DS_CNN/DS_CNN_L.pb and why I can't read variable from Pretrained_models/DS_CNN/DS_CNN_L.pb?

Thank you!

navsuda, 2018-06-22:
Hi @WeldonWangwang,
The dimensions for the model DS_CNN_L.pb are in table-7 in the [paper](https://arxiv.org/pdf/1711.07128.pdf). It is possible that the variable names might have changed since the model was trained first (apologies for that confusion). You can print all the variables in a model pb file using the following:
for var in tf.get_collection(tf.GraphKeys.GLOBAL_VARIABLES)
  print(var.name)
But in general, if you want to quantize the model to deploy on MCU, I suggest you start from a checkpoint.

navsuda, 2018-07-11:
Hi @WeldonWangwang,

Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda