ARM-software/ML-KWS-for-MCU #27: conver into tensorflow lite

cogmeta, 2018-05-02:
Can someone help me in converting the frozen graph model to tensorflow lite? thanks.

cogmeta, 2018-05-02:
Traceback (most recent call last):
  File "/home/ubuntu/exp/tensorflow-lite/tensorflow/tensorflow/python/tools/freeze_graph.py", line 382, in <module>
    run_main()
  File "/home/ubuntu/exp/tensorflow-lite/tensorflow/tensorflow/python/tools/freeze_graph.py", line 379, in run_main
    app.run(main=my_main, argv=[sys.argv[0]] + unparsed)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/platform/app.py", line 48, in run
    _sys.exit(main(_sys.argv[:1] + flags_passthrough))
  File "/home/ubuntu/exp/tensorflow-lite/tensorflow/tensorflow/python/tools/freeze_graph.py", line 378, in <lambda>
    my_main = lambda unused_args: main(unused_args, flags)
  File "/home/ubuntu/exp/tensorflow-lite/tensorflow/tensorflow/python/tools/freeze_graph.py", line 272, in main
    flags.saved_model_tags, checkpoint_version)
  File "/home/ubuntu/exp/tensorflow-lite/tensorflow/tensorflow/python/tools/freeze_graph.py", line 254, in freeze_graph
    checkpoint_version=checkpoint_version)
  File "/home/ubuntu/exp/tensorflow-lite/tensorflow/tensorflow/python/tools/freeze_graph.py", line 99, in freeze_graph_with_def_protos
    _ = importer.import_graph_def(input_graph_def, name="")
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/framework/importer.py", line 285, in import_graph_def
    raise ValueError('No op named %s in defined operations.' % node.op)
ValueError: No op named DecodeWav in defined operations.

cogmeta, 2018-05-02:
I solved by stripping the model of DecodeWav and now  i am able to convert to tflite.

closed 2018-05-02 by cogmeta

bhchoi, 2018-12-12:
> I stripped the model of DecodeWav operator and now i am able to convert to tflite.
Hello @cogmeta. I have same problem.
Could you tell me how to strip DecodeWav operator and parameter value of freeze_graph.py?