ARM-software/ML-KWS-for-MCU #28: Couldn't find build tools in your program.

zhuanaa, 2018-05-03:
zhuan@zhuan-HP:~/zhuan/ML-KWS-for-MCU/Deployment$ mbed new kws_simple_test --mbedlib 
[mbed] Creating new program "kws_simple_test" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at latest revision in the current branch
[mbed] Unpacking library build "5aab5a7997ee" in "/home/zhuan/zhuan/ML-KWS-for-MCU/Deployment/kws_simple_test/mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/5aab5a7997ee"
[mbed] ## **Couldn't find build tools in your program.** Downloading the mbed 2.0 SDK tools...

anybody comes across this problem "Couldn't find build tools in your program."? Look forward to your comment, thank you!

navsuda, 2018-05-07:
@zhuanaa 
Its not an error, its just a message that it couldn't find tools and hence downloading them.  See [this](https://github.com/ARMmbed/mbed-cli#creating-a-new-program-for-mbed-os-2)

navsuda, 2018-07-11:
Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda