ARM-software/ML-KWS-for-MCU #29: The accuracy is not very high

sunjunlishi, 2018-05-07:
is not 16000hz，16bit，1 channel？

sunjunlishi, 2018-05-11:
oh，i found it is very good。
if train more and more,It must be commercially available

sunjunlishi, 2018-05-15:
When you train, how many voice files do you have in a word? If I add voice files, the training results will be more generalized.

navsuda, 2018-05-22:
Hi @sunjunlishi,
This dataset consists of ~2500 per word, for exact numbers please see [this paper](https://arxiv.org/pdf/1804.03209.pdf) or check your extracted tar of the dataset.  Yes, detection accuracy  will improve if you add more meaningful data. 
Please consider closing the issue if you don't have any follow up questions.

sunjunlishi, 2018-05-23:
thanks for your answer。’A‘ ’AO‘ will be recognized as  ‘off’，i think Adding an unknown sample，will be ok？

closed 2018-05-23 by sunjunlishi

ccnankai, 2018-11-23:
@sunjunlishi 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?