ARM-software/ML-KWS-for-MCU #30: Extract weight from pb

z430, 2018-05-10:
Hi, thanks for sharing your code.
how you can get the ds_cnn_weights.h value? 
Please tell me how to get those value,

thx

navsuda, 2018-05-22:
Hi @z430,
You have to follow the same steps as the DNN example: first quantize weights and then layer input/outputs and call the appropriate CMSIS-NN functions.

navsuda, 2018-07-11:
Updated the quant_models.py to help in quantizing the DS_CNN models. 

Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda

ccnankai, 2018-11-23:
@z430 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?