ARM-software/ML-KWS-for-MCU #31: Extract weight from checkpoint

cam510, 2018-05-11:
Hi,
I found I can run quant_test.py whit dnn model, 
but I run quant_test.py whit ds_cnn model has a error.(copy the models method create_ds_cnn_model to quant_models)
the error row is quant_test.py -> np.savetxt(f,var_values.transpose(),fmt='%d',delimiter=', ',newline=', ')
how can I fix it?

navsuda, 2018-06-22:
Hi @cam510 ,
The error was because of the dimensions of the convolution weights. I have added the support for convolution layers in [quant_test.py](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/quant_test.py#L110).  Also added ds_cnn model in quant_models for your reference.

cam510, 2018-06-25:
@navsuda thanks very much

closed 2018-07-11 by navsuda