ARM-software/ML-KWS-for-MCU #32: This error shows that "....might take a minute", however it cost one hour and still donot success.

oneway3124, 2018-05-14:
C:\Users\Administrator\Desktop\ML-KWS-for-MCU\Deployment>mbed new kws_simple_tes
t --mbedlib
[mbed] Creating new program "kws_simple_test" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbe
d/builds" at latest revision in the current branch
[mbed] Downloading library build "5aab5a7997ee" (might take a minute)

oneway3124, 2018-05-14:
C:\Users\Administrator\Desktop\ML-KWS-for-MCU\Deployment>mbed new kws_simple_tes
t --mbedlib
[mbed] Creating new program "kws_simple_test" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbe
d/builds" at latest revision in the current branch
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/m
bed/builds/tip"
warning: LF will be replaced by CRLF in mbed.bld.
The file will have its original line endings in your working directory.
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK t
ools...

oneway3124, 2018-05-14:
C:\Users\Administrator\Desktop\ML-KWS-for-MCU\Deployment>cd kws_simple_test

C:\Users\Administrator\Desktop\ML-KWS-for-MCU\Deployment\kws_simple_test>mbed de
ploy
[mbed] Updating library "mbed" to branch/tag "tip"
[mbed] Downloading library build "5aab5a7997ee" (might take a minute)

navsuda, 2018-05-14:
This seems to be an issue with your mbed-cli environment. Can you try importing and building mbed-cli example [here](https://github.com/ARMmbed/mbed-cli#importing-an-existing-program) and raise an issue in mbed-cli if you have the same problem?

oneway3124, 2018-05-18:
[image]

[image]

[image]

From the above pictures, I think that this issue may be created by package update. So Can you tell me which file is responsible for this?

oneway3124, 2018-05-18:
The second picture is my local file storage which is compiled for latest successful compilation.

oneway3124, 2018-05-22:
@navsuda Can you help me answer this question?

oneway3124, 2018-05-22:
It seems to be ok now!

[image]

But, why is there some other compile error as the figure shows?

navsuda, 2018-05-22:
Hi @oneway3124 
I do not understand the cause of your error, but it seems to be an error with mbed-cli specific to your environment. Can you try debugging it in a virtual-environment? Also, raise an issue in mbed-cli, some other users who have had this issue may be able to help you. 
I have added an example for FRDM-K64F, which can be built without mbed-cli [here](https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment/Examples/simple_test_k64f_gcc). Using that as a starting point, you may be able to build it for any other mbed supported board.

oneway3124, 2018-05-23:
OK, this environment has been installed on the laptop of my colleague, but the error looks like me.
Which virtual environment do you mean? virtualbox or vmware, I has tried it.
At the same time, I also raise one issue in mbed-cli, there is no useful answer for this question.
So, I am trying your new example for FRDM-K64F now.

oneway3124, 2018-05-23:
[image]
@navsuda thanks for your help, I can compile the FRDM-K64F on the ubuntu system in the virtualbox.
Then, I will complete the remaining steps of NUCLEO_F411RE board. If there is some question, I will raise some isssue on this site. Thanks very much in advance.

closed 2018-05-23 by oneway3124

ccnankai, 2018-11-23:
@oneway3124 

In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?

ccnankai, 2018-11-23:
@oneway3124  
你好，量化的时候   --act_max参数的值怎么确定的，从哪得来的  是1-128吗？