ARM-software/ML-KWS-for-MCU #33: Wondered about stm32 supported board list.

leejaeuk, 2018-05-18:
Hello, I'm very thank you for ARM team's effort.

I have several questions about stm32 support board.
We already have several stm32 board(stm32f051, f103rb, f407ig)

But these boards are not word in mbed-cli environments. (not supported)
So, we purchase other boards to applicated mbed-cli. 

1. their is another way that work in my stm32(ML-KWS)? 

2. Also i have seen other developer repo, that page implement cortex-m4 board KWS.
then, is it possible to applies ARM-KWS any other board that supported mbed-cli?

Thank you for your support.

navsuda, 2018-06-07:
Hi @leejaeuk,
You can run the provided examples on any mbed-cli supported boards (as long as the board has sufficient memory). You can also potentially run the provided examples on boards that are not supported on mbed-cli. An example with Makefile is provided here: https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment/Examples/simple_test_k64f_gcc
Hope it helps.

leejaeuk, 2018-06-08:
ok, it is late for me, but thanks to your help.

closed 2018-06-08 by leejaeuk

ccnankai, 2018-11-23:
@leejaeuk 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?

leejaeuk, 2018-11-23:
@ccnankai 
Sorry, actually i didn't execute any quant_test.py beacause other works busy :(

ccnankai, 2018-11-23:
@leejaeuk   
thank you very much