ARM-software/ML-KWS-for-MCU #34: The mbed-cli error leading to the compiling error.

oneway3124, 2018-05-23:
Hi author,

I has tried so many method for this compiling issue. 
When I has done sucessfully mbed deploy , then mbed compile, some error shows as following.

[image]

[image]

the error log says, ....mbed.h: No such file or directory  ... compilation terminated.

But I use `find name mbed.h`
only shows
`./mbed/5aab5a7997ee/mbed.h`

So I can draw a conclusion the mbed-cli lead to the compiling error. So why will we use the previous version of mbed-cli.

@navsuda certainly, we also can continue to the gcc-arm toolchain development.
At the same time , we should keep the mbed-cli toolchain development.  What's your opinion?

navsuda, 2018-06-22:
@oneway3124 ,
Sorry, I didn't get your question correctly. 
Both toolchains serve different sets of developer communities and hence are important. Please use the toolchain which works best for your purpose.

navsuda, 2018-07-11:
Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda

ccnankai, 2018-11-27:
@oneway3124 
I've had the same problem.is it a version problem. how can I solve this problem?