ARM-software/ML-KWS-for-MCU #35: GCC Comile Issue

gamnes, 2018-05-28:
Hi, I am having problems compiling your application using the newly added GCC "option". The error i get is the following after following your steps (clone CMSIS_5 into repo, cd into gcc project, then make);
"""
...GIT\ML-KWS-for-MCU\Deployment\Examples\simple_test_k64f_gcc>make -j 8
make[1]: *** No rule to make target 'simple_test/main.o', needed by 'simple_test_k64f_gcc.elf'.  Stop.
Makefile:25: recipe for target 'all' failed
make: *** [all] Error 2
"""

Any thoughts as to why this is not working? Based on another issue, it seems they have gotten this to work already so just curious as to what i am doing wrong.

leejaeuk, 2018-06-11:
I have same problem.. 
i trying to virtual environment (Ubuntu) but not working.

navsuda, 2018-06-13:
It seems you haven't installed the GNU Arm Embedded toolchain on your machine: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads. If you haven't, then download the latest version in the above link, unzip it and add its path to your $PATH environment variable.

gamnes, 2018-06-13:
Hi navsuda, Thank you! I will try to make gnu arm available from PATH and see if that helps!:) I'll update this issue if so..

Another question I had, how would you approach running the real-time example on a slower running processor then the one you have tested with? I think you mentioned in your article that the algorithm runs in 17.something ms on the device you were testing. If the device is less powerful, how do you alter the algorithm? Do you have to change the weights - which means you will have to rerun the whole training i suppose? It seems the call into DS_CNN::run_nn() will take the same time regardless of the size of in_data, so if that function takes 200ms on some device, i assume either the input size will have to change, or the algorithm inside run_nn() itself?

navsuda, 2018-07-11:
Closing the issue due to inactivity, please reopen it if you still face the issue.

closed 2018-07-11 by navsuda