ARM-software/ML-KWS-for-MCU #37: Issue with MFCC banks noise in realtime test

reklessabandon, 2018-06-03:
When I run the code, unmodified, for KWS inference on live audio on STM32F746NG development kit, I get a solid band which hugely negatively impacts the accuracy. This band exists even when audio.IN.SetVolume() is set down to 20. I can't hear any noise when listening through the line out. The band also exists when the audio input is changed from the onboard MEMs mic to Line 1, so it is not an issue with the microphone itself. A photo is attached when the volume is lowered down to 20. I would really appreciate any ideas to eliminate this issue.
[image]

navsuda, 2018-06-22:
@reklessabandon,
does it mean it isn't detecting any words at all? 
The figure looks normal to me, the first band is just the DC component of MFCC and since TensorFlow MFCC Op scales down the input audio by (2^16), the first MFCC component shows as negative value. You can cross check this with your tensorflow MFCC results on the dataset.

reklessabandon, 2018-07-03:
The attached picture shows the correct classification, but that is misleading. The solid first band results in no correct word detections. 
The image in the paper "Hello Edge: Keyword Spotting on Microcontrollers", shows a dark green band in this position, suggesting that my board is displaying the opposite as a baseline.

navsuda, 2018-07-11:
@reklessabandon,
The image shown in the paper uses an older version of mfcc routine, please do not compare it with yours. 
Does it mean yours isn't detecting any words at all? If so, there is some problem with your board or your compilation flow. The best way to debug it is to compare the MFCC output from Tensorflow with yours by printing it out.

ccnankai, 2018-11-23:
@gm-jl 
ow to get the value of the quantization ranges -- act_max when run quant_test.py

closed 2018-12-03 by navsuda