ARM-software/ML-KWS-for-MCU #39: Extracting weights and quantizatuin for DSCNN

pooyaww, 2018-06-14:
Hi,
I see that, currently, **quant_test.py** only supports extracting and quantizing of DNN weights.
Is it possible that you share a script that also supports **ds_cnn**  ?

Thanks you

navsuda, 2018-06-22:
@pooyaww ,
Added ds-cnn in quant_test.py. However you would have to fold the batch-norm parameters to the preceding convolution layer weights/biases using fold_batchnorm.py script, which generates a different checkpoint with updated weights/biases. I will add some notes on this in Quantization guide. 

@leejaeuk,
glad to hear that you tested DS-CNN on several boards in real-time !!! I'm sure DS-CNN model detection accuracy in the real-world may be improved by more training with more background noise.

pooyaww, 2018-06-25:
@navsuda Thank you for sharing your code. I will try it.
As I understood by your explanation, after quantizing weights/biases and activation data by quant_test.py, I should call  fold_batchnorm.py to fold batch-norm parameters and generate updated weights/biases. But, at this step, considering that a new checkpoint is generated, don't we need to quantize activation data again ? Will the scaling factors remain the same?

navsuda, 2018-07-11:
Closing this issue, please reopen it if you still face an error.

closed 2018-07-11 by navsuda

ccnankai, 2018-11-22:
@pooyaww   
Follow your steps ，use fold_batchnorm.py to generate a new  model.  Then execute the 
 quant_test.py ，but quant_test.py  requires parameter --act_max , How is the value of this parameter obtained?
 After quant_test.py is executed, is the quantization complete?