ARM-software/ML-KWS-for-MCU #40: Demo experiences compile time error when using mbed-os 5

mbartling, 2018-06-16:
Hello,

I experience the following compile time error when working with mbed-os 5. I am pretty sure it's because mbed5 is running CMSIS4 and not CMSIS5. Are there any plans to update this demo to support mbed OS 5?

`__STATIC_FORCEINLINE is undefined` <- not available in CMSIS4

navsuda, 2018-06-21:
Hi @mbartling,
I couldn't replicate this error with mbed-os 5, can you point me to what you are doing?
If you explicitly point to CMSIS-5 when compiling as shown in [this example](https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment#build-and-run-a-simple-kws-inference), it shouldn't give the error you mentioned.

navsuda, 2018-07-11:
@mbartling,
Can you confirm if compiling explicitly with CMSIS-5 helped?

ccnankai, 2018-11-23:
@mbartling 
how to set the value of the quantization ranges -- act_max when run quant_test.py

closed 2018-12-03 by navsuda