ARM-software/ML-KWS-for-MCU #43: realtime_test compiling steps fail with "mbed tools not found"

4p0pt0Z, 2018-06-25:
Hi,

I can not get the realtime_test example to work following the steps [here](https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment).

`mbed new kws_realtime_test --create-only`
produces the warning
`[mbed] WARNING: Cannot find the mbed tools directory in`...

The warning is repeated in the output of the command `mbed deploy`

Finally, `mbed compile ...` fails with
`[mbed] ERROR: The mbed tools were not found in ...`
`[mbed] ERROR: Run `mbed deploy` to install dependencies and tools.`

Is this a [naming problem](https://github.com/ARMmbed/mbed-cli/issues/296) ? (mbed instead of mbed-os ?) or a [new behavior](https://github.com/ARMmbed/mbed-cli/issues/646) of mbed-cli ?

Thanks

4p0pt0Z, 2018-06-26:
Hi navsuda,

I have no file "requirements.txt" in the mbed folder in my kws_realtime_test directory. If you are talking about [this file](https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt), then I have all the requirements installed.

I have been able to make it work by creating the project directory without the `--create-only` option, and then _not_ copying the mbed.lib file from the Example/realtime_test/mbed_libs folder. 

Thanks for your help

closed 2018-06-26 by 4p0pt0Z

ccnankai, 2018-11-23:
@4p0pt0Z 

In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set  the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?

eera-l, 2019-07-02:
Hi, 
I ran into the same problem as @4p0pt0Z (the mbed tools were not found) and tried to make it work by creating the directory without `--create-only` and by not copying the mbed.lib file. 
Deploying worked, but now that I try to compile I get the error:

`[Fatal Error] kws_f746ng.h@22,10: AUDIO_DISCO_F746NG.h: No such file or directory
[ERROR] In file included from ../Examples/realtime_test/main.cpp:25:
../Source/KWS_F746NG/kws_f746ng.h:22:10: fatal error: AUDIO_DISCO_F746NG.h: No such file or directory
`
because deploying in that way did not create any AUDIO_DISCO_F746NG.h files. 
Do you have any idea on how to solve this problem?
Thanks

eera-l, 2019-07-02:
I solved the problem by:
1. creating the kws_realtime_test folder as described here: [https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment](url)
2. at this point I tried to deploy, but ran into "mbed tools not found error", so I added mbed-os to kws_realtime_test with `mbed add mbed-os`
3. removed the mbed folder from kws_realtime_test
4. deployed
5. at this point I tried to compile, but ran into the issue described here: [https://github.com/ARM-software/ML-KWS-for-MCU/issues/94](url). So I checked out the older version of CMSIS_5, as stated in the page, and I was finally able to compile.

Bosch936, 2021-09-03:
Hi. 

I have the same problem and i've trie to solve like you. When i copy the .lib into my folder and then run mbed deploy it works fine but then give me an error on the following command. The following code is a part of the message:

      |     ^~~~~~~~~
./mbed-os/events/include/events/UserAllocatedEvent.h:310:5: note: C++11 'constexpr' only available with '-std=c++11' or '-std=gnu++11'
./mbed-os/events/include/events/UserAllocatedEvent.h:317:5: error: 'constexpr' does not name a type
  317 |     constexpr UserAllocatedEvent(const T *obj, R(T::*method)(ArgTs...) const, ArgTs... args) :
      |     ^~~~~~~~~
./mbed-os/events/include/events/UserAllocatedEvent.h:317:5: note: C++11 'constexpr' only available with '-std=c++11' or '-std=gnu++11'
./mbed-os/events/include/events/UserAllocatedEvent.h:324:5: error: 'constexpr' does not name a type
  324 |     constexpr UserAllocatedEvent(EventQueue *q, const T *obj, R(T::*method)(ArgTs...) const, ArgTs... args) :
      |     ^~~~~~~~~
./mbed-os/events/include/events/UserAllocatedEvent.h:324:5: note: C++11 'constexpr' only available with '-std=c++11' or '-std=gnu++11'
./mbed-os/events/include/events/UserAllocatedEvent.h:331:5: error: 'constexpr' does not name a type
  331 |     constexpr UserAllocatedEvent(volatile T *obj, R(T::*method)(ArgTs...) volatile, ArgTs... args) :
      |     ^~~~~~~~~
./mbed-os/events/include/events/UserAllocatedEvent.h:331:5: note: C++11 'constexpr' only available with '-std=c++11' or '-std=gnu++11'
./mbed-os/events/include/events/UserAllocatedEvent.h:338:5: error: 'constexpr' does not name a type
  338 |     constexpr UserAllocatedEvent(EventQueue *q, volatile T *obj, R(T::*method)(ArgTs...) volatile, ArgTs... args) :
      |     ^~~~~~~~~
./mbed-os/events/include/events/UserAllocatedEvent.h:338:5: note: C++11 'constexpr' only available with '-std=c++11' or '-std=gnu++11'
./mbed-os/events/include/events/UserAllocatedEvent.h:345:5: error: 'constexpr' does not name a type
  345 |     constexpr UserAllocatedEvent(const volatile T *obj, R(T::*method)(ArgTs...) const volatile, ArgTs... args) :
      |     ^~~~~~~~~
./mbed-os/events/include/events/UserAllocatedEvent.h:345:5: note: C++11 'constexpr' only available with '-std=c++11' or '-std=gnu++11'
./mbed-os/events/include/events/UserAllocatedEvent.h:352:5: error: 'constexpr' does not name a type
  352 |     constexpr UserAllocatedEvent(EventQueue *q, const volatile T *obj, R(T::*method)(ArgTs...) const volatile, ArgTs... args) :
      |     ^~~~~~~~~
./mbed-os/events/include/events/UserAllocatedEvent.h:352:5: note: C++11 'constexpr' only available with '-std=c++11' or '-std=gnu++11'
./mbed-os/events/include/events/UserAllocatedEvent.h: In static member function 'static void events::UserAllocatedEvent<F, void(ArgTs ...)>::event_dtor(void*)':
./mbed-os/events/include/events/UserAllocatedEvent.h:287:41: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
  287 |         UserAllocatedEvent<F, void(ArgTs...)> *instance = (UserAllocatedEvent<F, void(ArgTs...)> *)(((equeue_event *)p) - 1);
      |                                         ^~~
./mbed-os/events/include/events/UserAllocatedEvent.h:287:92: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
  287 |         UserAllocatedEvent<F, void(ArgTs...)> *instance = (UserAllocatedEvent<F, void(ArgTs...)> *)(((equeue_event *)p) - 1);
      |                                                                                            ^~~
./mbed-os/events/include/events/UserAllocatedEvent.h: At global scope:
./mbed-os/events/include/events/UserAllocatedEvent.h:358:31: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
  358 | template <typename F, typename... ArgTs>
      |                               ^~~
./mbed-os/events/include/events/UserAllocatedEvent.h:359:33: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
  359 | UserAllocatedEvent<F, void(ArgTs...)> EventQueue::make_user_allocated_event(F f, ArgTs... args)
      |                                 ^~~
./mbed-os/events/include/events/UserAllocatedEvent.h:359:91: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
  359 | UserAllocatedEvent<F, void(ArgTs...)> EventQueue::make_user_allocated_event(F f, ArgTs... args)
      |                                                                                           ^~~~
./mbed-os/events/include/events/UserAllocatedEvent.h: In member function 'events::UserAllocatedEvent<F, void(ArgTs ...)> events::EventQueue::make_user_allocated_event(F, ArgTs ...)':
./mbed-os/events/include/events/UserAllocatedEvent.h:361:44: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
  361 |     return UserAllocatedEvent<F, void(ArgTs...)>(this, f, args...);
      |                                            ^~~
./mbed-os/events/include/events/UserAllocatedEvent.h: At global scope:
./mbed-os/events/include/events/UserAllocatedEvent.h:364:43: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
  364 | template <typename T, typename R, typename... ArgTs>
      |                                           ^~~
./mbed-os/events/include/events/UserAllocatedEvent.h:365:45: warning: variadic templates only available with '-std=c++11' or '-std=gnu++11'
[... cut ...]
       Command: "/usr/bin/python2 -u /home/alessandro/Scrivania/ML-KWS/Deployment/kws_realtime_test/mbed-os/tools/make.py -t GCC_ARM -m DISCO_F746NG --profile ../release_O3.json --source . --source ../Source --source ../Examples/realtime_test --source ../CMSIS_5/CMSIS/NN/Include --source ../CMSIS_5/CMSIS/NN/Source --source ../CMSIS_5/CMSIS/DSP/Include --source ../CMSIS_5/CMSIS/DSP/Source --source ../CMSIS_5/CMSIS/Core/Include --build ./BUILD/DISCO_F746NG/GCC_ARM-RELEASE_O3 -j 8"
       Tip: You could retry the last command with "-v" flag for verbose output
---
How it is possible to compile with the oldest version of the mbed and not with the new one mbed-os?!