ARM-software/ML-KWS-for-MCU #45: Training other words

leejaeuk, 2018-06-26:
Hello, @navsuda 
I'm constantly development KWS application. and I tried several test in my stm32, mbed borad.
finally, I want to training my own words(except :  no, yes, rigth, left....)  like turn on, turn off, light on.

After reading your source code, I realized adding data_dir -> my_wav file. but is it only thing i have to do? To adding my own words, how can i do?
 or only using speech_commends_v0.02 data set, just adding word like' happy', 'house' then KWS is work well?

To product weight.h file, DNN process follow train.py -> test.py->quant_test.py -> weight.h
but i want to DS-CNN process. not sure about process 
i guess, fold_batchnorm.py-> test.py -> quant_test.py -> weight.h(from DS-CNN)
Am i right? (maybe wrong..)

Thank you.

pooyaww, 2018-06-26:
-just take care of other parameters available in the parser of training script. such as sample rate, clip duration and addressing to the directory
-You are right but you may use test.py only for checking the accuracy.

leejaeuk, 2018-06-27:
@pooyaww ,
Thank you for your reply. 
but i don't understand this sentence(' -> -just take care of other parameters available in the parser of training script. such as sample rate, clip duration and addressing to the directory')

If i want to adding new word, insert data_dir and prepare my wav data(like light on wavefile)
. Next when i execute train.py, you mean i considered wav file's sample rate, clip duration?

Thank you.

navsuda, 2018-07-11:
[This tutorial](https://www.tensorflow.org/tutorials/sequences/audio_recognition) has the best info on how to train for new words. 
Please make sure that you have a large-enough sample size for any new wav recordings that you add (>1000 utterances per word).

leejaeuk, 2018-07-17:
@navsuda 
Okay, i understand 
thank you!

closed 2018-07-17 by leejaeuk

ccnankai, 2018-11-23:
@leejaeuk 

In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py .
how to set  the specific value of the parameter act_max ( x 0 0 0 0 0 0 0 0 0 0 0 ) ?