ARM-software/ML-KWS-for-MCU #47: provided DS-CNN accuracy (on board)

4p0pt0Z, 2018-06-29:
Hi,

When testing the provided DS-CNN on my board, the accuracy seems to be very low. Especially in the real time example, the model almost always predicts "left" as soon as there is some sound. 
Everything seems fine listening to the audio loopback, and the DNN provides much better results.
Is there specific parameters to use with DS-CNN in order to get high accuracy ?

Thanks

leejaeuk, 2018-07-02:
In my case, DS-CNN provided very high accuracy on my FRDM-K64 board.(both simple and real-time test are good) 
what is your board model name? 
I think is relate to microphone problem like wave file frequency or any other format.

leejaeuk, 2018-07-03:
is it any problem? when you compile using mbed-os? You should check a process of mbed compile.
before execute mbed new kws_realtime_test --create-only command, you must execute 

-cd Deployment
-git clone https://github.com/ARM-software/CMSIS_5.git 
-pip install mbed-cli

or you should using virtual machine ubuntu 16.04 version
and you can use mbed compile tool.

leejaeuk, 2018-07-03:
maybe you can modify Deployment/Examples/realtime_test/main.cpp file to 
Deployment/Examples/simple_test/main.cpp file. 
how to change DNN -> DS-CNN in your code?

4p0pt0Z, 2018-07-03:
The simple_test example works alright (predicts "right" with 99% confidence).
To change from DNN to DS-CNN? I modified 
- the include in "kws_f746ng.h"
- the parent class of KWS_F746NG in "kws_f746ng.h" and in the constructor implementation in "kws_f746ng.cpp"

The dataset is composed of close talk recorded samples, however there are far field on the board. Could it be that the DNN is simply more robust to ambient noise than the DS-CNN ?

sherylll, 2018-07-06:
Hi @4p0pt0Z ,
where did you get the value 99%? From what I understand the results `kws.output` are values of q7_t type, i.e. from -128 to 127?

Using an XMC4700 board (Cortex-M4), with DNN I got 127 for "right" and 0 for every word else. With DS-CNN I got 85 for silence and 42 for "right". Anyone having insights on this?

4p0pt0Z, 2018-07-06:
Hi @sherylll ,

> where did you get the value 99%? From what I understand the results kws.output are values of q7_t type, i.e. from -128 to 127?

Indeed, but you can for instance translate this value to percentage with `(int)kws.output[max_ind]*100/128` as is done in kws_simple_test/main.cpp

In my case, I get 127 for "right" with both DNN and DS-CNN, on a STM32F746G and a STM32F767ZI board.

sherylll, 2018-07-06:
Hi @4p0pt0Z 

Thanks a lot! I realized that I missed one line so I deleted my last comment... 

I also noticed that the scaled MFCC can exceed -127 in the simple test example, as I set a break point here: https://github.com/ARM-software/ML-KWS-for-MCU/blob/e3f1685b406c0cc2c5724d2278e8bf720b009cce/Deployment/Source/MFCC/mfcc.cpp#L210 

Is DNN robust against such cases, or is my board misbehaving?

4p0pt0Z, 2018-09-24:
Seems to be fixed by b6e13d6976c098cb6a564384b4c80a2c387d27b5

closed 2018-09-24 by 4p0pt0Z

ccnankai, 2018-11-23:
@4p0pt0Z 
how to get the value of the quantization ranges -- act_max when run quant_test.py

ccnankai, 2019-03-22:
@4p0pt0Z 
have encountered the same problem.  the model almost always predicts "down" as soon as there is some sound. you solved it?

4p0pt0Z, 2019-03-22:
Hi @ccnankai,
My problem got solved by b6e13d6976c098cb6a564384b4c80a2c387d27b5. Which version do you use ?