ARM-software/ML-KWS-for-MCU #49: Quant_guide.md example command failed

varteev, 2018-07-10:
I want to run quantization as described in Quant_guide.md. I run command as specified in Quant_guide.md. But I get error:
$ python quant_test.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --checkpoint Pretrained_models/DNN/DNN_S.pb 
/home/arteev/.local/lib/python2.7/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
2018-07-10 20:11:37.959657: I tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2
Traceback (most recent call last):
  File "quant_test.py", line 308, in <module>
    tf.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/platform/app.py", line 126, in run
    _sys.exit(main(argv))
  File "quant_test.py", line 206, in main
    FLAGS.model_architecture, FLAGS.model_size_info)
  File "quant_test.py", line 79, in run_quant_inference
    is_training=False)
  File "/home/arteev/work/src/ML-KWS-for-MCU/quant_models.py", line 115, in create_model
    act_max, is_training)
  File "/home/arteev/work/src/ML-KWS-for-MCU/quant_models.py", line 163, in create_dnn_model
    if(act_max[i]!=0):
IndexError: list index out of range

Does Quant_guide.md contain correct commands example? It seems command --model_size_info 144 144 144 is not correct.

navsuda, 2018-07-11:
@varteev,
1) "--checkpoint" should be the output of final checkpoint that you get after you run train.py as per the commands (and not the .pb file). 
2) You also missed the "--act_max" parameter as described in the Quant_guide.md. That will give the ranges for the quantization ranges for the activation data (or intermediate data).

varteev, 2018-07-11:
@navsuda ,

Thank you for quick response. After I run train as proposed from train_commands.txt:
python train.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10,10,10 --summaries_dir work/DNN10/DNN1/retrain_logs --train_dir work/DNN10/DNN1/training
I have following file generated
$ ls -l work/DNN10/DNN1/training/best/
-rw-r--r-- 1 arteev arteev     89 Jul 11 13:54 checkpoint
-rw-r--r-- 1 arteev arteev 955744 Jul 11 13:54 dnn_1734.ckpt-30.data-00000-of-00001
-rw-r--r-- 1 arteev arteev    910 Jul 11 13:54 dnn_1734.ckpt-30.index
-rw-r--r-- 1 arteev arteev 132847 Jul 11 13:54 dnn_1734.ckpt-30.meta

There is no single .ckpt file any more according to Tensorflow docs. 

But when I run quant_test, it failed. But the error is different in this case:
$ python quant_test.py --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40  --window_stride_ms 40 --checkpoint work/DNN10/DNN1/training/best/dnn_1734.ckpt-30.index --act_max 32 0 0 0 0
/home/arteev/.local/lib/python2.7/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
2018-07-11 14:09:10.580595: I tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2
INFO:tensorflow:Restoring parameters from work/DNN10/DNN1/training/best/dnn_1734.ckpt-30.index
Traceback (most recent call last):
  File "quant_test.py", line 308, in <module>
    tf.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/platform/app.py", line 126, in run
    _sys.exit(main(argv))
  File "quant_test.py", line 206, in main
    FLAGS.model_architecture, FLAGS.model_size_info)
  File "quant_test.py", line 90, in run_quant_inference
    models.load_variables_from_checkpoint(sess, FLAGS.checkpoint)
  File "/home/arteev/work/src/ML-KWS-for-MCU/quant_models.py", line 135, in load_variables_from_checkpoint
    saver.restore(sess, start_checkpoint)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/training/saver.py", line 1802, in restore
    {self.saver_def.filename_tensor_name: save_path})
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/client/session.py", line 900, in run
    run_metadata_ptr)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/client/session.py", line 1135, in _run
    feed_dict_tensor, options, run_metadata)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/client/session.py", line 1316, in _do_run
    run_metadata)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/client/session.py", line 1335, in _do_call
    raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.NotFoundError: Tensor name "Variable" not found in checkpoint files work/DNN10/DNN1/training/best/dnn_1734.ckpt-30.index
         [[Node: save/RestoreV2 = RestoreV2[dtypes=[DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT], _device="/job:localhost/replica:0/task:0/device:CPU:0"](_arg_save/Const_0_0, save/RestoreV2/tensor_names, save/RestoreV2/shape_and_slices)]]

Caused by op u'save/RestoreV2', defined at:
  File "quant_test.py", line 308, in <module>
    tf.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/platform/app.py", line 126, in run
    _sys.exit(main(argv))
  File "quant_test.py", line 206, in main
    FLAGS.model_architecture, FLAGS.model_size_info)
  File "quant_test.py", line 90, in run_quant_inference
    models.load_variables_from_checkpoint(sess, FLAGS.checkpoint)
  File "/home/arteev/work/src/ML-KWS-for-MCU/quant_models.py", line 134, in load_variables_from_checkpoint
    saver = tf.train.Saver(tf.global_variables())
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/training/saver.py", line 1338, in __init__
    self.build()
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/training/saver.py", line 1347, in build
    self._build(self._filename, build_save=True, build_restore=True)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/training/saver.py", line 1384, in _build
    build_save=build_save, build_restore=build_restore)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/training/saver.py", line 835, in _build_internal
    restore_sequentially, reshape)
  File "/home/arteev/tmp/ML-KWS-for-MCU-virtualenv-python27/local/lib/python2.7/site-packages/tensorflow/python/training/saver.py", line 472, in _AddRestoreOps
    restore_sequentially)
[... cut ...] ve/RestoreV2/tensor_names, save/RestoreV2/shape_and_slices)]]

"I tried different filed for --checkpoint (meta, data, checkpoint), but script failes with other error: Unable to open table file work/DNN10/DNN1/training/best/dnn_1734.ckpt-30.data-00000-of-00001: Data loss: not an sstable (bad magic number): perhaps your file is in a different file format and you need to use a different restore operator?"

I suppose I pass incorrect checkpoint file. But I can't understand which file script expects.

navsuda, 2018-07-11:
` --checkpoint work/DNN10/DNN1/training/best/dnn_1734.ckpt-30` should help.

varteev, 2018-07-11:
that helped! Thank you!

varteev, 2018-07-11:
I have one more question: after quant_test.py finish I can see that number of bits for fractional part for fixed point are 9 or even 14:
 
final_fc_0 number of wts/bias: (144, 12) dec bits: 9 max: (0.20117188,0.20042063) min: (-0.19726562,-0.19753574)
Variable_0 number of wts/bias: (12,) dec bits: 14 max: (0.0051879883,0.0051957406) min: (-0.002319336,-0.0023462817)

But it is supposed that total amount of available bits is 8 (because we calculate dec_bits = 7-int_bits). It seems that something is wrong. Is it correct that dec bits = 9? Or I just have to use those values as information for retrieving final output result values?

closed 2018-07-12 by varteev