ARM-software/ML-KWS-for-MCU #51: ask help about how to do quantization and use arm_fully_connected_q7?

placebokkk, 2018-07-12:
Hi,

Thanks for your nice work!

I am using arm_fully_connected_q7 for speeding my cortex m4 project. But the output is different with that using raw C implementation or arm_matrix_mul_f32.

I do the quantization by below steps:

> 1. Get min, max for Wk,Bk,Input_k,Ouput_k.  (Input_k = Output_{k-1})
> 2.calculate Q for Wk, Bk, Ik,Ok . 
> 3. calculate Bias Shift and output shift
> -  Bias_shift = Q_{Ik} + Q_{Wk}  - Q_{Bk} 
> -  Output_shift = Q_{Ik} + Q_{Wk}  - Q_{Ok} 
> 4. Quantize the Wk an Bk by the Q_{Wk} ,Q_{Bk} 
> 5. Use bias_shift, output_shift  in arm_fully_connected_q7()

I also try to print the de-quantized value for W and B at runtime and that is same with the raw float value.

Do I miss something? Do I need to reorder the elements of weight?  such as [1,2,3,4] to [1,3,2,4].

Below is my quantization params.

> #I use cmvm to normalize the fbank feature so it is in [-2.6,3.6]
> Q_i =5
> Q_W = [5, 6, 6, 7, 6]
> Q_B =[4, 4, 4, 5, 5]
> Q_O =[2, 1, 2, 2, 3]
> Shift_B = [6, 4, 3, 4, 3]
> Shift_O  =[8, 7, 5, 7, 5]

Thank you very much for helping!

placebokkk, 2018-07-13:
Sorry for bothering. I find the reason. 

The arm_fully_connected_q7 implement W*x.

I used to use X*W. 

After change the row-major W to col-major order, it works very well for me!

closed 2018-07-13 by placebokkk

saichand07, 2019-03-25:
@placebokkk 
could you tell me where did you change the code for speeding your cortex M4  (in the quantization process,) ??

Thanks

saichand07, 2019-05-03:
@placebokkk 

> Sorry for bothering. I find the reason.
> 
> The arm_fully_connected_q7 implements W_x. But I misunderstood it for X_W.
> 
> After changing the row-major W to col-major order, it works very well for me!

Hai @placebokkk 
could you please tell me, how did you change row-major W to col-major order.
my model is not detecting words correctly at all.

Thank you.

saichand07, 2019-05-09:
@placebokkk Thanks for rising this issue,

Could you please tell me, how did you change row-major W to col-major order ? can you give me brief explanation,

Thank you very much for your time

weishu20, 2019-06-24:
> Sorry for bothering. I find the reason.
> 
> The arm_fully_connected_q7 implements W_x. But I misunderstood it for X_W.
> 
> After changing the row-major W to col-major order, it works very well for me!

can you tell me how to calculate Q_i and Q_out,thank you very much!