ARM-software/ML-KWS-for-MCU #52: quant_test returns incorrect number of fraq bits?

varteev, 2018-07-12:
after quant_test.py finish I can see that number of bits for fractional part for fixed point are 9 or even 14:

final_fc_0 number of wts/bias: (144, 12) dec bits: 9 max: (0.20117188,0.20042063) min: (-0.19726562,-0.19753574)
Variable_0 number of wts/bias: (12,) dec bits: 14 max: (0.0051879883,0.0051957406) min: (-0.002319336,-0.0023462817)

It seems it is a bug. If wts/bias value is in range -1,1, number of integer bits should be 0, isn't it? Now according to formula int(log2(value)) number of bits for integer part become negative (because log of value <1 os negative). This cause number of fraq bits > 7.

varteev, 2018-07-13:
I suppose I misunderstand Q notation usage here. As I understand dec bits = 9 doesn't mean we need 9 bits for that. It is used to convert quanted values to real or to other Q formats.

closed 2018-07-13 by varteev

ccnankai, 2018-11-22:
Hi, @varteev 
 how to get the value of the quantization ranges -- act_max  when run  quant_test.py

saichand07, 2019-05-08:
Hello @varteev 
Did you solve Q notation problem?