ARM-software/ML-KWS-for-MCU #56: Examples\realtime_test\main.cpp line:79&95

xujieuse, 2018-07-21:
`arm_copy_q7((q7_t *)kws->audio_buffer, (q7_t *)(kws->audio_buffer)+2*(kws->audio_buffer_size-(kws->frame_len-kws->frame_shift)), 2*(kws->frame_len-kws->frame_shift));`

I have some doubts about this code.According to the comments(**//copy the last (frame_len - frame_shift) audio data to the start**) I think it should look like this.

`memmove( (q7_t *)(kws->audio_buffer)+2*(kws->audio_buffer_size-(kws->frame_len-kws->frame_shift)), (q7_t *)kws->audio_buffer,2*(kws->frame_len-kws->frame_shift));`

xujieuse, 2018-07-21:
I got it wrong

xujieuse, 2018-07-24:
@navsuda 
 ML-KWS-for-MCU/Deployment/Source/KWS/kws.cpp line:93
`//shift right old predictions`
`arm_copy_q7((q7_t *)predictions, (q7_t *)(predictions+num_out_classes), (sliding_window_len-1)*num_out_classes);`
predictions[0]->[num_out_classes] will overwrite predictions[1]->[num_out_classes],and then
predictions[1]->[num_out_classes] will overwrite predictions[2]->[num_out_classes]
...
It leads to all of predictions being the same as the first one.

navsuda, 2018-07-24:
@xujieuse,
Good catch in both cases !!
Fixed the error with audio_buffer copying in your first comment (which would only affect the DS_CNN model). 
Second one with sliding window of predictions can be fixed either by memmove or arm_copy_q7 by moving the window left (instead of right in the current version). Will push a fix for this too.

closed 2018-12-03 by navsuda