ARM-software/ML-KWS-for-MCU #63: Errors when running freeze.py

ColinNordin, 2018-10-05:
After having trained a ds_cnn using the following command:

`python3 train.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 100,100,100 --summaries_dir work/DS_CNN/DS_CNN1/retrain_logs --train_dir work/DS_CNN/DS_CNN1/training --data_dir ~/ML-KWS-for-MCU/speech_dataset --eval_step_interval 50`

I try to freeze the model using `freeze.py`, but I encounter some issues. Inside `train_dir/best` I can find a file named `checkpoint` as well as three different files for each actual checkpoint (e.g. : `ds_cnn_4965.ckpt-300.data-00000-of-00001`, `ds_cnn_4965.ckpt-300.index` and `ds_cnn_4965.ckpt-300.meta`)

Which of those should be used as `checkpoint` when running `freeze.py`?

I get some different errors depending on which of the three files I use as `checkpoint`.
1. When running: `python3 freeze.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --checkpoint ~/Documents/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300.data-00000-of-00001 --output_file ds_cnn.pb` I get the error: 

2018-10-05 16:49:17.586925: W tensorflow/core/util/tensor_slice_reader.cc:95] Could not open ~/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300.data-00000-of-00001: Data loss: not an sstable (bad magic number): perhaps your file is in a different file format and you need to use a different restore operator?
2018-10-05 16:49:17.588021: W tensorflow/core/util/tensor_slice_reader.cc:95] Could not open ~/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300.data-00000-of-00001: Data loss: not an sstable (bad magic number): perhaps your file is in a different file format and you need to use a different restore operator?
2018-10-05 16:49:17.588485: W tensorflow/core/framework/op_kernel.cc:1275] OP_REQUIRES failed at save_restore_tensor.cc:170 : Data loss: Unable to open table file ~/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300.data-00000-of-00001: Data loss: not an sstable (bad magic number): perhaps your file is in a different file format and you need to use a different restore operator?
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1278, in _do_call
    return fn(*args)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1263, in _run_fn
    options, feed_dict, fetch_list, target_list, run_metadata)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1350, in _call_tf_sessionrun
    run_metadata)
tensorflow.python.framework.errors_impl.DataLossError: Unable to open table file ~/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300.data-00000-of-00001: Data loss: not an sstable (bad magic number): perhaps your file is in a different file format and you need to use a different restore operator?
	 [[Node: save/RestoreV2 = RestoreV2[dtypes=[DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, ..., DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT], _device="/job:localhost/replica:0/task:0/device:CPU:0"](_arg_save/Const_0_0, save/RestoreV2/tensor_names, save/RestoreV2/shape_and_slices)]]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "freeze.py", line 193, in <module>
    tf.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/platform/app.py", line 125, in run
    _sys.exit(main(argv))
  File "freeze.py", line 124, in main
    models.load_variables_from_checkpoint(sess, FLAGS.checkpoint)
  File "~/ML-KWS-for-MCU/models.py", line 160, in load_variables_from_checkpoint
    saver.restore(sess, start_checkpoint)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 1725, in restore
    {self.saver_def.filename_tensor_name: save_path})
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 877, in run
    run_metadata_ptr)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1100, in _run
    feed_dict_tensor, options, run_metadata)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1272, in _do_run
    run_metadata)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1291, in _do_call
    raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.DataLossError: Unable to open table file ~/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300.data-00000-of-00001: Data loss: not an sstable (bad magic number): perhaps your file is in a different file format and you need to use a different restore operator?
	 [[Node: save/RestoreV2 = RestoreV2[dtypes=[DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, ..., DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT], _device="/job:localhost/replica:0/task:0/device:CPU:0"](_arg_save/Const_0_0, save/RestoreV2/tensor_names, save/RestoreV2/shape_and_slices)]]

Caused by op 'save/RestoreV2', defined at:
  File "freeze.py", line 193, in <module>
    tf.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/platform/app.py", line 125, in run
    _sys.exit(main(argv))
  File "freeze.py", line 124, in main
    models.load_variables_from_checkpoint(sess, FLAGS.checkpoint)
  File "~/ML-KWS-for-MCU/models.py", line 159, in load_variables_from_checkpoint
    saver = tf.train.Saver(tf.global_variables())
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 1281, in __init__
    self.build()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 1293, in build
    self._build(self._filename, build_save=True, build_restore=True)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 1330, in _build
    build_save=build_save, build_restore=build_restore)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 778, in _build_internal
    restore_sequentially, reshape)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 397, in _AddRestoreOps
    restore_sequentially)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 829, in bulk_restore
    return io_ops.restore_v2(filename_tensor, names, slices, dtypes)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/ops/gen_io_ops.py", line 1463, in restore_v2
[... 193 lines cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 1293, in build
    self._build(self._filename, build_save=True, build_restore=True)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 1330, in _build
    build_save=build_save, build_restore=build_restore)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 778, in _build_internal
    restore_sequentially, reshape)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 397, in _AddRestoreOps
    restore_sequentially)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tensorflow/python/training/saver.py", line 829, in bulk_restore
[... cut ...]
DataLossError (see above for traceback): Unable to open table file ~/Documents/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300.meta: Data loss: not an sstable (bad magic number): perhaps your file is in a different file format and you need to use a different restore operator?
	 [[Node: save/RestoreV2 = RestoreV2[dtypes=[DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, ..., DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT, DT_FLOAT], _device="/job:localhost/replica:0/task:0/device:CPU:0"](_arg_save/Const_0_0, save/RestoreV2/tensor_names, save/RestoreV2/shape_and_slices)]]

What am I doing wrong?

gamnes, 2018-10-08:
Hi, not an expert; but pretty sure you're only supposed to use the name of the checkpoint, 'ds_cnn_4965.ckpt-300' (of course including the path to the checkpoint folder, but don't use .data..., .index, or .meta). 
So you're not suppose to use any of the three files specifically, just the checkpoint name: {PATH_TO_CHECKPOINTS_FOLDER}/ds_cnn_4965.ckpt-300
In your case: 
--checkpoint ~/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300

The different checkpoints that you can use should be stored in a 'checkpoint' file in the same folder.

navsuda, 2018-10-08:
@ColinNordin,
You should provide checkpoint argument as `--checkpoint ~/Documents/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_4965.ckpt-300` for the freeze.py script.

closed 2018-12-03 by navsuda