ARM-software/ML-KWS-for-MCU #65: error run quant_test.py

OptimusPrime0, 2018-10-24:
all results classify  to  the first column. why is it ?

INFO:tensorflow:Confusion Matrix:
 [[1842    0    0    0    0    0    0    0    0    0    0    0]
 [1850    0    0    0    0    0    0    0    0    0    0    0]
 [1848    0    0    0    0    0    0    0    0    0    0    0]
 [1932    0    0    0    0    0    0    0    0    0    0    0]
 [1832    0    0    0    0    0    0    0    0    0    0    0]
 [1894    0    0    0    0    0    0    0    0    0    0    0]
 [1831    0    0    0    0    0    0    0    0    0    0    0]
 [1826    0    0    0    0    0    0    0    0    0    0    0]
 [1874    0    0    0    0    0    0    0    0    0    0    0]
 [1774    0    0    0    0    0    0    0    0    0    0    0]
 [1906    0    0    0    0    0    0    0    0    0    0    0]
 [1836    0    0    0    0    0    0    0    0    0    0    0]]
INFO:tensorflow:Training accuracy = 8.28% (N=22245)
INFO:tensorflow:set_size=3093
INFO:tensorflow:Confusion Matrix:
 [[258   0   0   0   0   0   0   0   0   0   0   0]
 [258   0   0   0   0   0   0   0   0   0   0   0]
 [261   0   0   0   0   0   0   0   0   0   0   0]
 [270   0   0   0   0   0   0   0   0   0   0   0]
 [260   0   0   0   0   0   0   0   0   0   0   0]
 [264   0   0   0   0   0   0   0   0   0   0   0]
 [247   0   0   0   0   0   0   0   0   0   0   0]
 [256   0   0   0   0   0   0   0   0   0   0   0]
 [257   0   0   0   0   0   0   0   0   0   0   0]
 [256   0   0   0   0   0   0   0   0   0   0   0]
 [246   0   0   0   0   0   0   0   0   0   0   0]
 [260   0   0   0   0   0   0   0   0   0   0   0]]
INFO:tensorflow:Validation accuracy = 8.34% (N=3093)
INFO:tensorflow:set_size=3081
INFO:tensorflow:Confusion Matrix:
 [[257   0   0   0   0   0   0   0   0   0   0   0]
 [257   0   0   0   0   0   0   0   0   0   0   0]
 [256   0   0   0   0   0   0   0   0   0   0   0]
 [252   0   0   0   0   0   0   0   0   0   0   0]
 [272   0   0   0   0   0   0   0   0   0   0   0]
 [253   0   0   0   0   0   0   0   0   0   0   0]
 [267   0   0   0   0   0   0   0   0   0   0   0]
 [259   0   0   0   0   0   0   0   0   0   0   0]
 [246   0   0   0   0   0   0   0   0   0   0   0]
 [262   0   0   0   0   0   0   0   0   0   0   0]
 [249   0   0   0   0   0   0   0   0   0   0   0]
 [251   0   0   0   0   0   0   0   0   0   0   0]]
INFO:tensorflow:Test accuracy = 8.34% (N=3081)

navsuda, 2018-12-03:
That happens when you train the model with one set of hyperparameters (window_size, window_stride, etc.) and test it with a different set.

closed 2018-12-03 by navsuda

Haryslee, 2019-07-27:
Hi, I have the same problem, did you solved it as navsuda said?