ARM-software/ML-KWS-for-MCU #68: act_max parameter for ds-cnn model

leiming0225, 2018-11-08:
@navsuda
another issue is, when I run python quant_test.py --act_max 64 0 0 0 0 0 0 0 0 0 0 0 .... for ds-cnn model,
I can get a training accuracy which is very close to the training accuracy output by test.py.
but when I try to quantizing the second parameter by command:
python quant_test.py --act_max 64 x 0 0 0 0 0 0 0 0 0 0
I tried replace x with 128,64,32,16,8,4,2,1
all output training accuracy is much lower than when x==0
could you please give some idea for this case? thanks

ccnankai, 2018-11-23:
@leiming0225 
In the quantization process, how to get the value of the quantization ranges -- act_max in quant_test.py . 
Where do you get the specific value of the parameter act_max       64 0 0 0 0 0 0 0 0 0 0 0  ?

ccnankai, 2018-11-26:
@leiming0225 
I've had the same problem. how can I solve this problem?

wangtiance, 2018-11-26:
My understanding is that if you set the nth value to 0, the model won't do any quantization at layer n. if the value is >0, the activation function at layer n is quantized to [-val, val) with 8 bit precision, causing a great accuracy drop since we don't know a priori what a suitable range looks like. This quantization range is supposed to be a learnable parameter, but I don't think it works that way here.

ccnankai, 2018-11-27:
@pooyaww 
Admire your  ability ,thank you for your explanation .

wayne175, 2019-04-14:
> @navsuda
> another issue is, when I run python quant_test.py --act_max 64 0 0 0 0 0 0 0 0 0 0 0 .... for ds-cnn model,
> I can get a training accuracy which is very close to the training accuracy output [...]

I've had the same problem. how can I solve this problem?