ARM-software/ML-KWS-for-MCU #71: How to convert these models to TF-Lite models and run it

rayeh5, 2018-11-29:
I can't find useful information to convert these models to TF-Lite and run it at Linux environment.
Could someone give some hints?
Thanks!

pooyaww, 2018-11-29:
- Train your desired model
- Freez it by the provided python script ang generate a pb file.
- Use TensorFlow Lite Converter to convert .pb file to .tflite file
- Use it in TF-lite !
[check the documentation](https://www.tensorflow.org/lite/convert/)

rayeh5, 2018-11-30:
> * Train your desired model
> * Freez it by the provided python script ang generate a pb file.
> * Use TensorFlow Lite Converter to convert .pb file to .tflite file
> * Use it in TF-lite !
>   [check the documentation](https://www.tensorflow.org/lite/convert/)

Thanks for the reply.
I think that I don'e explain my question well.

I try to convert the Pretrained_models/CNN/CNN_S.pb via the following commands:
---------------------
toco \
   --input_file=./CNN_S.pb \
   --input_format=TENSORFLOW_GRAPHDEF \
   --output_format=TFLITE \
   --output_file=./CNN_S.tflite \
   --inference_type=FLOAT \
   --input_shape=1,49,10,1 \
   --input_array=x \
   --input_data_type=FLOAT \
   --output_arrays=final_fc
---------------------------------------
The following are the error messages:
From the error messages, it seems that:
1. TF-Lite doesn't support operation: "DecodeWav", "AudioSpectrogram" and "Mfcc";
2. I don't know how to assign the "output_arrays", where can find this information?
----------------------------------------------
Error messages:

N$ ./my_toco_float.sh 
2018-11-30 14:23:35.412174: I tensorflow/contrib/lite/toco/import_tensorflow.cc:1233] Converting unsupported operation: DecodeWav
2018-11-30 14:23:35.412606: I tensorflow/contrib/lite/toco/import_tensorflow.cc:1233] Converting unsupported operation: AudioSpectrogram
2018-11-30 14:23:35.412876: I tensorflow/contrib/lite/toco/import_tensorflow.cc:1233] Converting unsupported operation: Mfcc
2018-11-30 14:23:35.414064: F tensorflow/contrib/lite/toco/tooling_util.cc:711] Check failed: model.HasArray(output_array) Output array not found: final_fc
./my_toco_float.sh: line 10:  3577 Aborted                 (core dumped) toco --input_file=./CNN_S.pb --input_format=TENSORFLOW_GRAPHDEF --output_format=TFLITE --output_file=./CNN_S.tflite --inference_type=FLOAT --input_shape=1,49,10,1 --input_array=x --input_data_type=FLOAT --output_arrays=final_fc

---------------------------------

Thanks for your time.

pooyaww, 2018-12-04:
@rayeh5 Obviously it is an issue with toco which currently does not support all operations. The trained models here include some modules to extract PCM data from wave files, finding spectrum and computing MFCC.  As tensorflow has a good back-end for Audio recognition, I think you can open an issue in https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite and asking for a support regarding these operations.

marjanemd, 2019-08-20:
What are the input array and output arrays here? How are they defined exactly?