ARM-software/ML-KWS-for-MCU #83: Deployement of CRNN on STM32F746 mcu

saichand07, 2019-01-21:
@navsuda 
Thanks for this tutorial,
 I have a quantized version of CRNN (.py , pb) and its weights (.h), How can we deploy this python model on to the Microcontroller, Do we need to write hard code of CRNN (cpp, h).

Could you please provide me the answer for this problem.

bkanaki, 2019-01-25:
Hi @saichand07 , Great ob in quantizing the CRNN model. I have a quick question for you. Could you share some insights on how did you go about quantizing the CRNN? You don't need to share your code, but if you can give me a brief overview, it would be great!

Thanks a lot.

bkanaki, 2019-01-25:
Thank you, @saichand07 . I believe you used `fake_quant_with_min_max_vars` function, correct? The examples use that. But how did you calculate the act_max for each layer? OR did you just keep it at the default 128? I am not certain about how to calculate these values as of now. I checked the examples in `quant_models.py`.

Once again, thanks for your help.

bkanaki, 2019-01-25:
I have got an idea, one can plot the activation summary on Tensorboard and then get the max values for the activation layers. For someone who is stuck at the same point, I believe this will help.

saichand07, 2019-01-27:
yes, I have used fake quant twice and I got nearly 1% depletion in accuracy.
have you tried with DS_CNN (medium)model giving 95% test accuracy but after quant_test, it generated only 9% (test,train,valid acc) 
do you have any idea ??

bkanaki, 2019-01-28:
Thanks! Were you able to verify that the fake quant nodes are added after each activation in the both GRU cells? For example, by inspecting the frozen graph trough tensorboard.

saichand07, 2019-01-29:
Thanks for askingthis question :),
I had verified that the fake_quant nodes are added after each activation in the both GRU cells, 
Have you deployed any of these models on MCU boards ?

bkanaki, 2019-01-29:
No. Unfortunately I cannot verify the two fake quant nodes for each GRU cell. So I haven't done the next part.

HKLee2040, 2019-03-29:
@saichand07 
I'm new in this topic.
Would you mind to share the .py of CRNN to me?

saichand07, 2019-04-02:
@HKLee2040 
CRNN GRU, LSTM are provided in this tutorial, check models.py

HKLee2040, 2019-04-05:
@saichand07 
Got it!  Thanks