ARM-software/ML-KWS-for-MCU #84: Change of wanted_words resulting very low recognition accuracy (only 24%)

auvilink, 2019-01-30:
Hi, there, I changed several wanted_words in "train.py" as:
"up,down,left,right,stop,go,follow,forward,backward,on,off"

I then trained ds_cnn_s model using instructions in "train_commands.txt"
"python train.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DS_CNN/DS_CNN1/retrain_logs --train_dir work/DS_CNN/DS_CNN1/training"

I changed the wanted_words in "test.py" to the same ones above and 

"python test.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --checkpoint=/Users/michaelhe/ml/asr/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_9294.ckpt-27200"

**INFO:tensorflow:Test accuracy = 24.29% (N=4492)**

24.29% accuracy seems very low.

Did I make any mistake somewhere in the process?

Thank you so much.

saichand07, 2019-01-30:
Hey @auvilink , 
How did you create the wanted_words (.wav files) and Have you followed the properties of speech commands dataset V_0.0.2 ??
properties like size, frequency, sample_rate, clip_duration, background_noise, etc.
There are many factors that influenced your model accuracy. 
if possible could you share that .wav files that you created ??

auvilink, 2019-01-30:
@saichand07 
Hi, there, thanks for the reply.
All the words / .wav files are in original speech commands dataset V_0.0.2. 
I did not use my own wav files.

Thanks again.

saichand07, 2019-01-30:
@auvilink 
thanks for the answer, 
Have you trained your model again after adding 'forward, backward' words to the wanted words list ??

saichand07, 2019-01-30:
@auvilink 
you need to train your model again after adding those new words to the wanted_words list, later you can test it by using test.py 
it will work

auvilink, 2019-01-30:
@saichand07 
I DID train it again using following command:
"python train.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DS_CNN/DS_CNN1/retrain_logs --train_dir work/DS_CNN/DS_CNN1/training"

The training process took almost a full day. 

That's how I got all the model trained checkpoints saved in "work/DS_CNN/DS_CNN1/training/best/" directory.

**### Could you please kindly enlighten me where did I made mistake or maybe the right way to retrain the model?**

As matter of fact, I tried this procedure three times. But all with low recognition test accuracy.

Thanks.

saichand07, 2019-01-31:
@auvilink don't worry, I will try with my models and i'll let you know the information

saichand07, 2019-01-31:
hai @yanyanem 
Have you deployed any models on to the MCU (STM32F746NG), could you please give me short description about it, I tried as per instructions in this tutorial but its not working in my case.
it would be really helpful if i could get any info on it.
thanks

auvilink, 2019-02-01:
@yanyanem @saichand07 
Thanks guys, it works well now. 
Really appreciated.

 [[375   0   0   0   0   0   0   0   0   0   0   0   0]
 [  2 248   4  26  12  11   6  35   3  10   4   9   5]
 [  3  10 371   1  10   1   7   3   1   0   2   2  14]
 [  4  14   1 373   3   0   0   8   1   0   1   0   1]
 [  2  17   4   2 382   1   2   0   0   0   2   0   0]
 [  2  28   2   0   6 352   2   0   0   0   0   1   3]
 [  2   4   6   2   0   1 386   4   0   0   0   0   6]
 [  5  28   4  29   1   0   4 325   3   0   0   0   3]
 [  0   5   0   0   1   0   5   9 127  15   0   8   2]
 [  3  13   0   1   0   0   0   6   6 115   5   3   3]
 [  2  23   1   0   5   2   1   0   0   0 131   0   0]
 [  4  12   4   4   0   0   0   1   7   1   0 342  21]
 [  1   2  19   0   5   0   2   6   4   2   0   9 352]]
INFO:tensorflow:Test accuracy = 86.35% (N=4492)

closed 2019-02-01 by auvilink

saichand07, 2019-02-04:
Hello @yanyanem 
It would be really helpful if I could get an answer from you,

D:\KWS-for-MCU\Deployment\kws_realtime_test>mbed compile -m DISCO_F746NG -t GCC_ARM \ --source . --source ..\Source --source ..\Examples\realtime_test / --source ..\CMSIS_5\CMSIS\NN\Include --source ..\CMSIS_5\CMSIS\NN\Source / --source ..\CMSIS_5\CMSIS\DSP\Include --source ..\CMSIS_5\CMSIS\DSP\Source / --source ..\CMSIS_5\CMSIS\Core\Include / --profile ..\release_O3.json -j 8
[mbed] Working path "D:\KWS-for-MCU\Deployment\kws_realtime_test" (program)
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
usage: make.py [-h] [-m MCU] [--custom-targets CUSTOM_TARGETS_DIRECTORY]
               [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG]
               [-p PROGRAM | -n PROGRAM | -L | -S [{matrix,toolchains,targets}]]
               [-j JOBS] [-v] [--silent] [-D MACROS] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [--ignore IGNORE] [-b BAUD] [--rpc] [--usb] [--dsp] [--testlib]
               [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: unrecognized arguments: \ / / / /
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 2
       Path: "D:\KWS-for-MCU\Deployment\kws_realtime_test"
       Command: "c:\python27\python.exe -u D:\KWS-for-MCU\Deployment\kws_realtime_test\mbed-os\tools\make.py -t GCC_ARM -m DISCO_F746NG --profile ..\release_O3.json --source . --source ..\Source --source ..\Examples\realtime_test --source ..\CMSIS_5\CMSIS\NN\Include --source ..\CMSIS_5\CMSIS\NN\Source --source ..\CMSIS_5\CMSIS\DSP\Include --source ..\CMSIS_5\CMSIS\DSP\Source --source ..\CMSIS_5\CMSIS\Core\Include --build .\BUILD\DISCO_F746NG\GCC_ARM-RELEASE_O3 \ / / / / -j 8"
       Tip: You could retry the last command with "-v" flag for verbose output
---

I can't able to fix this problem.

Thank you very much