ARM-software/ML-KWS-for-MCU #86: WARNING: MBED_ARM_PATH set as environment variable but doesn't exist

saichand07, 2019-02-05:
Hello @yanyanem @navsuda 
It would be really helpful if I could get an answer from you,

## D:\KWS-for-MCU\Deployment\kws_realtime_test>mbed compile -m DISCO_F746NG -t GCC_ARM \ --source . --source ..\Source --source ..\Examples\realtime_test / --source ..\CMSIS_5\CMSIS\NN\Include --source ..\CMSIS_5\CMSIS\NN\Source / --source ..\CMSIS_5\CMSIS\DSP\Include --source ..\CMSIS_5\CMSIS\DSP\Source / --source ..\CMSIS_5\CMSIS\Core\Include / --profile ..\release_O3.json -j 8
[mbed] Working path "D:\KWS-for-MCU\Deployment\kws_realtime_test" (program)
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
usage: make.py [-h] [-m MCU] [--custom-targets CUSTOM_TARGETS_DIRECTORY]
[-t TOOLCHAIN] [--color] [--cflags CFLAGS]
[--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
[--profile PROFILE] [--app-config APP_CONFIG]
[-p PROGRAM | -n PROGRAM | -L | -S [{matrix,toolchains,targets}]]
[-j JOBS] [-v] [--silent] [-D MACROS] [-f GENERAL_FILTER_REGEX]
[--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
[--extra EXTRA] [--peripherals PERIPHERALS]
[--dep DEPENDENCIES] [--source SOURCE_DIR]
[--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
[--ignore IGNORE] [-b BAUD] [--rpc] [--usb] [--dsp] [--testlib]
[--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: unrecognized arguments: \ / / / /
[mbed] ERROR: "c:\python27\python.exe" returned error.
Code: 2
Path: "D:\KWS-for-MCU\Deployment\kws_realtime_test"
Command: "c:\python27\python.exe -u D:\KWS-for-MCU\Deployment\kws_realtime_test\mbed-os\tools\make.py -t GCC_ARM -m DISCO_F746NG --profile ..\release_O3.json --source . --source ..\Source --source ..\Examples\realtime_test --source ..\CMSIS_5\CMSIS\NN\Include --source ..\CMSIS_5\CMSIS\NN\Source --source ..\CMSIS_5\CMSIS\DSP\Include --source ..\CMSIS_5\CMSIS\DSP\Source --source ..\CMSIS_5\CMSIS\Core\Include --build .\BUILD\DISCO_F746NG\GCC_ARM-RELEASE_O3 \ / / / / -j 8"
Tip: You could retry the last command with "-v" flag for verbose output
I can't able to fix this problem.

Thank you very much

yanyanem, 2019-02-11:
@saichand07 
i do not see this error before. but i can share u my steps. 

1. setup python2.7, and pip mbed. 
2. download gcc-arm-none-eabi-6-2017-q2-update and extract to my local and add PATH to .bashrc
export PATH="/home/ai/eric/tools/gcc-arm-none-eabi-6-2017-q2-update/bin:$PATH"

by the way, you can try kws_simple_test firstly then try kws_realtime_test next.

saichand07, 2019-02-11:
Thank you very much @yanyanem

saichand07, 2019-02-13:
@yanyanem 
Thanks for the answer
kws_simple_test is working properly and it is generating .bin and .elf files, But kws_realtime_test is not working , I have followed same tool chain, it is generating below error

**[mbed] ERROR: The mbed tools were not found "/home/***/ML-KWS-FOR-MCU/Deployment/kws_realtime_test".
you can run "mbed deploy" to install dependencies and tools.**

I run the mbed deploy several times but its still showing same error, and I also install "mbed-os" to solve this, then its generating new error like ''tools in mbed directory is not unique ''

could you explain me how to deal with this problem,

Thanks
saichand

saichand07, 2019-02-22:
Hai @yanyanem 
could you tell me what kind of output can we expect from kws_simple_test ??
Does it always display same output_class or does it always display some random output_class ??

closed 2019-03-04 by saichand07