ARM-software/ML-KWS-for-MCU #89: Memory sizes of the MFCC variables

saichand07, 2019-02-18:
Hello @navsuda @yanyanem

I want to change deployment source code (dynamic memory allocation to static memory allocation), It would be really helpful if I could get memory sizes of the MFCC variables.

Thanks and regards
Saichand

bkanakietac, 2019-02-23:
Hey, did you figure out @saichand07 ?

The MFCC parameters are computed [here](https://github.com/ARM-software/ML-KWS-for-MCU/blob/8151349b110f4d1c194c085fcc5b3535bdf7ce4a/models.py#L41). The variable `fingerprint_size` is of the interest.

Hope this helps. :)

saichand07, 2019-02-23:
Hai thanks for the reply,
I want to use static memory allocation for the deployment on MCUs, so I need to know the sizes of the parameters like,
FRAME_SHIFT ((int16_t)(SAMP_FREQ * 0.001 * FRAME_SHIFT_MS))
FRAME_LEN ((int16_t)(SAMP_FREQ * 0.001 * FRAME_LEN_MS))
scratch_pad = new q7_t[SCRATCH_BUFFER_SIZE];
 buffer1 = scratch_pad;
  buffer2 = buffer1 + (CONV1_OUT_CH*CONV1_OUT_X*CONV1_OUT_Y);
  col_buffer = buffer2 + (CONV2_OUT_CH*CONV2_OUT_X*CONV2_OUT_Y);

I calculated manually  from known parameters, but my debugger generating some stack underflow errors due to wrong buffer size allocations.
Do you know how to solve this problem??

bkanakietac, 2019-02-25:
What is SAMP_FREQ?

saichand07, 2019-02-25:
sampling rate 16000 of wav

closed 2019-03-01 by saichand07