ARM-software/ML-KWS-for-MCU #91: [Error] ds_cnn_weights.h@1,28988: too many initializers for 'const q7_t [2560] {aka const sign

lancezhangsf, 2019-03-04:
HI,
　I use my own data trainning model, then gen the weights.h,  i chang  arrays'name in weights.h and rename the weights.h for ds_cnn_weights.h
  compile the bin for stm32f746, f322cb7e/targets.json
[Error] @,: Compiler version mismatch: Have 5.4.1; expected version >= 6.0.0 and < 7.0.0
Building project kws_realtime_test (DISCO_F746NG, GCC_ARM)
Scan: kws_realtime_test
Scan: Source
Scan: realtime_test
Scan: Include
Scan: Source
Scan: Include
Scan: Source
Scan: Include
Compile [100.0%]: ds_cnn.cpp
[Error] ds_cnn_weights.h@1,28988: too many initializers for 'const q7_t [2560] {aka const signed char [2560]}'
[Error] ds_cnn_weights.h@2,717: too many initializers for 'const q7_t [64] {aka const signed char [64]}'
[Error] ds_cnn_weights.h@3,6690: too many initializers for 'const q7_t [576] {aka const signed char [576]}'
[Error] ds_cnn_weights.h@4,785: too many initializers for 'const q7_t [64] {aka const signed char [64]}'

saichand07, 2019-03-04:
@lancezhangsf 
I understand your problem,
 delete your current ds_cnn_weights.h, generate it again and then compile. 
don't overwrite the file again again, that adds additional weights to your existed ds_cnn_weights.h, so that size of your ds_cnn_weights.h will increase.

that's why your getting this [Error] ds_cnn_weights.h@1,28988: too many initializers for 'const q7_t [2560] {aka const signed char [2560]}'

saichand07, 2019-03-04:
@lancezhangsf 
could you tell me, how did you create your own dataset, did you do any data augmentation process ? 
how much data have you been collected?

Thanks
saichand

lancezhangsf, 2019-03-05:
> @lancezhangsf
> I understand your problem,
> delete your current ds_cnn_weights.h, generate it again and then compile.
> don't overwrite the file again again, that adds additional weights to your existed [...]

I  have resloved the error, because of the output_class overflow in main.cpp.

saichand07, 2019-03-05:
could you tell me, how did you create your own dataset, did you do any data augmentation process ?
how much data have you been collected?

Thanks
saichand

lancezhangsf, 2019-03-05:
> @lancezhangsf
> could you tell me, how did you create your own dataset, did you do any data augmentation process ?
> how much data have you been collected?
> 
> Thanks
> saichand

you can refer  https://www.tensorflow.org/tutorials/sequences/audio_recognition

saichand07, 2019-03-05:
Actually I want to create my own dataset with some special keywords, you recorded the samples with which kind of device and doest it require and post and pre processing ?

lancezhangsf, 2019-03-06:
> Actually I want to create my own data set with some special keywords, what kind of device have you used for recording samples and does it require any post and pre processing ?

Using stm32f746,  
No need to process data，just need to change the  wanted_words  in   train.py  and test.py.......

closed 2019-03-06 by lancezhangsf

saichand07, 2019-03-26:
@lancezhangsf 
I have resloved the error, because of the output_class overflow in main.cpp.

How did you resolved your error, what changes have you made in the code?