ARM-software/ML-KWS-for-MCU #92: How to quant dnn

barrycool, 2019-03-06:
Dear All,
    please help me! Can you tell me how to quan a dnn, python test.py works, but after I quan,  but it does not work on MCU, I get a wrong result
     I use below three commands:
python train.py --wanted_words yes,no --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --data_dir /datasets/kws --summaries_dir work/DNN/DNN1/retrain_logs --train_dir work/DNN/DNN1/training 

python fold_batchnorm.py --wanted_words yes,no --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --data_dir /datasets/kws --checkpoint work/DNN/DNN1/training/best/dnn_10000.ckpt-400

python quant_test.py --wanted_words yes,no --model_architecture dnn --model_size_info 144 144 144 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 40 --data_dir /datasets/kws --checkpoint work/DNN/DNN1/training/best/dnn_10000.ckpt-400_bnfused --act_max 32 0 0 0 0 0 0

saichand07, 2019-03-06:
Hai @barrycool 
could you tell me precisely what kind of error or result you got, which code have you used for the deployment on MCU

barrycool, 2019-03-06:
> Hai @barrycool
> could you tell me precisely what kind of error or result you got, which code have you used for the deployment on MCU

Hi @saichand07 below code, I used in stm32f429, when I run dnn_weights.h from ARM, I can get correct result, but after I collect myself dataset, train and test work well, test result is 97%, but after I quant it and run in stm32f429, I get wrong result, for example I input "right", then result is "unknown", so I want to know what command you use to train and quant dnn in your datasets
https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment

barrycool, 2019-03-14:
I get answers, please reference below
https://github.com/cledic/ML-KWS-for-MCU/blob/master/README2.md

closed 2019-03-14 by barrycool

ccnankai, 2019-03-22:
@barrycool 
I trained speech_commandsv2. Accuracy is 94% on the test set, but on stm32f7x, no keywords are detected.
The problems caused by the quantized weights.h lead to inaccurate predictions. May I ask how to correct it.

barrycool, 2019-03-23:
@ccnankai 
did you use yourself dataset? if not, please reference below, I can get correct result by micphone
https://github.com/cledic/ML-KWS-for-MCU/blob/master/README2.md

if you use yourself dataset, you must set same number of keywords, if number is different, I found I can't get correct result.
for example:
origin dataset: "yes", "no", "up", "down", "left", "right", "on", "off", "stop", "go"
your dataset: "cat", "dog", "hourse", "mouse", "left", "right", "on", "off", "stop", "go"