ARM-software/ML-KWS-for-MCU #93: the accuracy is very low when using my own data training models(on board)

lancezhangsf, 2019-03-07:
Hi,
    when using my own data for training model, and instand of the ds_cnn_weights.h ,
recompile the kws_realtime_test.bin ,but on board(stm32f746) ,it's difficult to recongize.  but i test the model on pc,
the accuracy is very high.

python test.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1    --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20  --checkpoint /opt/zsf_workspace/WorkSpace/SoundWorkSpace/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_9750.ckpt-30000
2019-03-07 14:18:00.541952: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2019-03-07 14:18:00.571174: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2499835000 Hz
2019-03-07 14:18:00.574544: I tensorflow/compiler/xla/service/service.cc:150] XLA service 0x41ee720 executing computations on platform Host. Devices:
2019-03-07 14:18:00.574588: I tensorflow/compiler/xla/service/service.cc:158]   StreamExecutor device (0): <undefined>, <undefined>
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/tensorflow/python/framework/op_def_library.py:263: colocate_with (from tensorflow.python.framework.ops) is deprecated and will be removed in a future version.
Instructions for updating:
Colocations handled automatically by placer.
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/confusion_matrix.py:193: to_int64 (from tensorflow.python.ops.math_ops) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.cast instead.
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/confusion_matrix.py:194: to_int32 (from tensorflow.python.ops.math_ops) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.cast instead.
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/tensorflow/python/training/saver.py:1266: checkpoint_exists (from tensorflow.python.training.checkpoint_management) is deprecated and will be removed in a future version.
Instructions for updating:
Use standard file APIs to check for files with this prefix.
INFO:tensorflow:Restoring parameters from /opt/zsf_workspace/WorkSpace/SoundWorkSpace/ML-KWS-for-MCU/work/DS_CNN/DS_CNN1/training/best/ds_cnn_9750.ckpt-30000
INFO:tensorflow:set_size=5686
INFO:tensorflow:Confusion Matrix:
 [[505   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0]
 [  0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0]
 [  0   0 512   0   0   0   0   0   0   0   0   0   0   0   0   0]
 [  0   0   0 259   0   0   0   0   0   0   0   0   0   0   0   0]
 [  0   0   0   0 260   0   0   0   0   0   0   0   0   0   0   0]
 [  0   0   0   0   0 263   0   0   0   0   0   0   0   0   0   0]
 [  0   0   0   0   0   0 233   0   0   0   0   0   0   0   0   0]
 [  0   0   0   0   0   0   0 512   0   0   0   0   0   0   0   0]
 [  0   0   0   0   0   0   0   0 343   0   0   0   0   0   0   0]
 [  0   0   0   0   0   0   0   0   0 529   0   0   0   0   0   0]
 [  0   0   0   0   0   0   0   0   0   0 253   0   0   0   0   0]
 [  0   0   0   0   0   0   0   0   0   0   0 454   0   0   0   0]
 [  0   0   0   0   0   0   0   0   0   0   0   0 283   0   0   0]
 [  0   0   0   0   0   0   0   0   0   0   0   0   0 383   0   0]
 [  0   0   0   0   0   0   0   0   0   0   0   0   0   0 521   0]
 [  0   0   0   0   0   0   0   0   0   0   0   0   0   0   0 376]]
INFO:tensorflow:Training accuracy = 100.00% (N=5686)
INFO:tensorflow:set_size=721
INFO:tensorflow:Confusion Matrix:
 [[66  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0 61  0  0  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0  0 31  1  0  0  0  0  0  1  0  0  0  0  0]
 [ 0  0  0  1 35  0  1  0  0  0  0  0  0  0  0  0]
 [ 0  0  2  0  2 28  0  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0  1  0 35  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0  0  0  0 74  0  0  0  0  0  0  0  1]
 [ 0  0  0  0  0  0  0  1 31  0  0  0  0  0  0  1]
 [ 0  0  1  0  0  0  0  0  0 53  0  0  0  0  0  0]
 [ 0  0  2  0  0  0  0  0  0  0 26  0  2  0  0  0]
 [ 0  0  0  0  0  2  0  0  0  0  0 70  0  0  0  0]
 [ 0  0  0  0  0  1  0  0  0  0  0  0 25  0  0  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0 47  0  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0 67  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 52]]
INFO:tensorflow:Validation accuracy = 97.23% (N=721)
INFO:tensorflow:set_size=729
INFO:tensorflow:Confusion Matrix:
 [[67  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0 59  0  0  0  1  0  0  0  0  0  0  1  0  0]
 [ 0  0  0 31  1  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0 19  0  1  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0  2 29  0  0  0  0  0  0  1  0  0  0]
 [ 0  0  0  0  1  0 39  0  0  0  0  0  0  0  0  0]
[... 81 lines cut ...]
 [ 0  0  0  0  0  3  0  0  0  0  0 69  0  0  0  0]
 [ 0  0  0  0  0  1  0  0  0  0  0  0 25  0  0  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0 47  0  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0 67  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 52]]
INFO:tensorflow:Validation accuracy = 87.93% (N=721)
INFO:tensorflow:set_size=729
INFO:tensorflow:Confusion Matrix:
 [[ 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 67]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0 57  0  0  0  1  0  0  0  0  0  0  3  0  0]
 [ 0  0  0 31  1  0  0  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0 19  0  1  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0  2 29  0  0  0  0  0  0  1  0  0  0]
 [ 0  0  0  0  1  0 39  0  0  0  0  0  0  0  0  0]
 [ 0  0  0  0  0  0  0 51  0  0  0  0  0  0  0  1]
 [ 0  0  0  0  0  0  0  1 56  0  0  0  0  0  0  0]
 [ 0  0  1  0  0  0  0  0  0 59  0  0  1  0  0  0]
 [ 0  0  0  0  0  0  0  0  0  2 27  0  0  0  0  0]
 [ 0  0  0  1  0  1  0  0  0  0  0 69  0  0  0  0]
 [ 0  0  0  0  0  0  0  0  0  4  0  0 38  0  0  0]
 [ 0  0  0  0  0  0  0  0  0  1  0  2  0 57  0  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0 54  0]
 [ 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 51]]
INFO:tensorflow:Test accuracy = 87.38% (N=729)

ccnankai, 2019-03-22:
@lancezhangsf 
I have encountered the same problem. Have you solved it?

bkanakietac, 2019-03-22:
in the quant_test.py, all the values of `--act_max 32 0 0 0 0 0 0 0 0 0 0 0` has to be some power of two instead of zeros. Check out the quant_guide and re-read the [quantization section](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Quant_guide.md#quantizing-kws-models) again. 

You need to calculate the bias shift and out shift for each layer and then put them in the cpp source code. It is not trivial, but definitely works. 

Also check #68 , #51 and try reproducing the simple DNN example first. We all want high accuracy results but if you figure out one simple, it would be easier to extend the idea to any model in this work.

ccnankai, 2019-03-23:
@bkanakietac 
sorry to bother you.I am a novice who does not understand quantification, I can't understand how to calculate. If act_max=32,4,64,16, 8, shift is calculated. Input 32, ouput 8, three layers of hidden layers.

ccnankai, 2019-03-24:
@bkanakietac  
I calculated the last layer, wx=8 bias=10  bias need to be shifted right 。 But there is no option for the right shift 。

bkanaki, 2019-03-24:
first of all, good that you figured some stuff. secondly, for the right shift, there is a problem. I had the same issue and logged #90 but no one responded. 

So input to your last layer is Q4.3 and weights Q1.6 but bias is Qx.10. Try changing this input layer to Q2.5 and see if the accuracy is still maintained (or maybe better?) and then your bias shift will be zero. I did something similar.

ccnankai, 2019-03-24:
@bkanaki 
Thank you for your kind help.

saichand07, 2019-04-02:
@bkanakietac @ccnankai 
Could you please tell me how to change the bias_shift and out_shift in cpp file
I have calculated the the dec_bits of every layer by using the quant_test.py,
I really do not understand, how to make use of dec_bits to adjust the bias (left or right)

It would be really helpful If some one could explain shortly with a example, I deployed my own model on to the board but it's always detecting one or two classes randomly,

In python version my model giving good accuracy but on board the word spotting is complete wrong,
I followed the all parameters from this tutorial.

saichand07, 2019-04-04:
> So input to your last layer is Q5.2 and weights Q1.6 but bias is Qx.10. Try changing this input layer to Q3.4 and see if the accuracy is still maintained (or maybe better?) and then your bias shift will be zero. I did something similar.

@bkanaki how did you figure out the Qm.n format of the layers and how to change the m,n values of the layer?

can you please explain ?  

Thanks

bkanakietac, 2019-04-04:
Read, experiment, repeat! Did you read the quantization guide above mentioned above? It is already decently documented with an example. All the relevant code is also in this repository. Explore!

As I had mentioned previously, it would be a good start to experiment with the DNN model because of its simplicity, but everyone just wants to use dscnn. 😒

wayne175, 2019-04-14:
> @bkanaki
> Thank you for your kind help.

I've had the same problem. how can I solve this problem?

wayne175, 2019-04-14:
> first of all, good that you figured some stuff. secondly, for the right shift, there is a problem. I had the same issue and logged #90 but no one responded.
> 
> So input to your last layer is Q5.2 and weights Q1.6 but bias is Qx.10. Try changing this input layer to Q3.4 and see if the accuracy is still maintained (or maybe better?) and then your bias shift will be zero. I did something similar.

I've had the same problem. how can I solve this problem?

saichand07, 2019-05-03:
> Read, experiment, repeat! Did you read the quantization guide above mentioned? It is already decently documented with an example. All the relevant code is also in this repository. Explore! The answer to your last question is in your previous comment.
> 
> As I had mentioned previously, it would be a good start to experiment with the DNN model because of its simplicity, but everyone just wants to use dscnn. 😒

@bkanakietac thank you for this comment,

I have calculated the bias shift and out_shift of each layer and I put them in cpp,the only  DNN model is working well but the DS_CNN model is not detecting the words properly at all,
I also tuned the recording window, averaging window len, detection threshold
if you have any idea, could you please tell me what still i need to consider.

JeffyCN, 2019-08-05:
> I've had the same problem. how can I solve this problem?

Seems like it's not supported:
https://github.com/ARM-software/CMSIS_5/issues/327#issuecomment-369690102

Adjust act_max might avoid that.