ARM-software/ML-KWS-for-MCU #94: multiple definitions of math functions in kws_simple_test

jeremyholleman, 2019-03-21:
I'm trying to compile the simple kws inference program described in the deployment page.  I get loads of "multiple definitions of XX" errors.  Here is the first one:
```
Link: arm-none-eabi-gcc @./BUILD/DISCO_L476VG/GCC_ARM-RELEASE_O3/.link_options.txt
[DEBUG] Return: 1
[DEBUG] Errors: BUILD/DISCO_L476VG/GCC_ARM-RELEASE_O3/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f32.o: In function `arm_abs_f32':
[DEBUG] Errors: /home/jeremy/ML-KWS-for-MCU/Deployment/kws_simple_test/../CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f32.c:102: multiple definition of `arm_abs_f32'
[DEBUG] Errors: BUILD/DISCO_L476VG/GCC_ARM-RELEASE_O3/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/BasicMathFunctions.o:/home/jeremy/ML-KWS-for-MCU/Deployment/kws_simple_test/../CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f32.c:102: first defined here
```
They are all under CMSIS_5/CMSIS/DSP/.  I don't see any obvious reason for multiple definitions. My compile command is 

`mbed compile -m DISCO_L476VG -t GCC_ARM --source .                --source ../Source/KWS        --source ../Source/NN       --source ../Source/MFCC                --source ../Source/local_NN       --source ../Examples/simple_test       --source ../CMSIS_5/CMSIS/NN/Include        --source ../CMSIS_5/CMSIS/NN/Source       --source ../CMSIS_5/CMSIS/DSP/Source               --source ../CMSIS_5/CMSIS/DSP/Include        --source ../CMSIS_5/CMSIS/Core/Include         --profile ../release_O3.json -j 8  -vv > compile.log`

Any ideas?  Is there a known good set of versions (of python, gcc, mbed) that I should try?

System details:
I've replaced the newest mbed library (65be27845400) with 3a7713b1edbc to solve the problem of a missing mbed.h file (as described [here](https://github.com/ARM-software/ML-KWS-for-MCU/issues/72)).  I'm using python 2.7 under anaconda. I'm running Ubuntu.  I've tried it with both the 8-2018-q4-major and the 6-2017-q2-update versions of the arm gcc compiler.

saichand07, 2019-03-21:
@jeremyholleman 
mbed compile -m DISCO_L476VG -t GCC_ARM --source . --source ../Source/KWS --source ../Source/NN --source ../Source/MFCC --source ../Source/local_NN --source ../Examples/simple_test --source ../CMSIS_5/CMSIS/NN/Include --source ../CMSIS_5/CMSIS/NN/Source --source ../CMSIS_5/CMSIS/DSP/Source --source ../CMSIS_5/CMSIS/DSP/Include --source ../CMSIS_5/CMSIS/Core/Include --profile ../release_O3.json -j 8
try this if not 
update your mbed-os repository

jeremyholleman, 2019-03-22:
Thanks @saichand07.  I tried your command line and got the same result.  It looks like the same command as is suggested on the project page; what I was using is just that plus the "very verbose" flag and redirecting the output to a file.

When you say to update the mbed-os repo, I'm not sure I understand.   The '--mbedlib' flag indicated on the README.md page causes it to get the library rather than the os.  The most recent version of the library causes an error due to missing mbed.h file (known issue, linked in OP.  Is there a workaround I missed?).  I tried the 'mbed new' command without the 'mbedlib' flag, so it fetched the (most recent version of) mbed-os instead of the library.  That gave me the same errors.

Am I missing anything?  Any other ideas?

saichand07, 2019-03-22:
Sloved by this:

mbed.h error :
download
https://os.mbed.com/users/mbed_official/code/mbed/archive/3a7713b1edbc.tar.gz
extract to ./mbed/ replace old one.

the old one lost mbed.h and platform folder , drivers folder.

jeremyholleman, 2019-03-22:
That solves the "missing mbed.h" problem.  I had switched the mbed library version to 3a77 to fix that.  But there is still the "multiple definitions" problem that you can see in the output listing of the original post.  It almost seems like .../BasicMathFunctions/arm_abs_f32.c is being included twice and generating a conflict with itself, but I can't figure out why that would be happening.  

If you (or anyone else) have an environment where this code is building correctly, could you provide the details of that, and I can try to reproduce it?

Thanks,
Jeremy

jeremyholleman, 2019-03-25:
The problem seems to be that within CMSIS_5, there are several "collection" c files, that #include other C files.   For example, 'DSP/Source/TransformFunctions/TransformFunctions.c' #includes arm_rfft_q31.c (and several other files). Then both arm_rfft_q31.c and TransformFunctions.c are compiled and linked, resulting in multiple definitions of the functions inside arm_rfft_q31.

Any ideas on (a) why is the build system including both the 'collection' files and the individual constituent files or (b) why doesn't this fail for everyone?  @navsuda ?   Again, if anyone can list the details of an environment where this code is building correctly ( mbed version, python version, toolchain and version, platform and os version) I could try to reproduce that.

Thanks
Jeremy

navsuda, 2019-03-25:
@jeremyholleman It failed for me too, but works fine with an [older commit](1d777eeeba686ba8e1f3e96c1f9a9396ae17ac1f) of CMSIS_5 where there is no [TransformFunctions.c](https://github.com/ARM-software/CMSIS_5/tree/1d777eeeba686ba8e1f3e96c1f9a9396ae17ac1f/CMSIS/DSP/Source/TransformFunctions).

jeremyholleman, 2019-03-25:
Thanks!  And yes.  I just tried it with an older commit.  It seems that TransformFunctions.c was checked in a couple weeks ago, along with a few other files that just include a bunch of other c files.  Those new files break the compile, at least with the way the mbed build function determines the list of C files to compile (essentially anything under the directories listed in the --source options, as far as I can tell).