ARM-software/ML-KWS-for-MCU #95: change wav file to 1500ms ,get poor result

ccnankai, 2019-03-21:
have you run ds_cnn on mcu already. I have only 200 training samples, the effect is very poor  on MCU. My wav file is 1500ms long.Can I run the program only by changing these three  parameters 
 FRAME_LEN_MS , FRAME_SHIFT_MS and  NUM_FRAMES? 
is it right？

saichand07, 2019-03-21:
Hi @ccnankai 
Did you create a dataset by on your own? 200 samples are not enough and did you follow the Speech command dataset properties (Ex: 16-bit pcm encoding of a wav with 16kHz sample rate)
Frame_len, frame_shift, num_frames, mel_freqs (lower and upper) etc are effects model accuracy. so you need to tune those parameters.

ccnankai, 2019-03-21:
@saichand07 
Thank you for your advice.
It seems that adjusting these five parameters is enough.right?
In the process of quantification, when adjusting act_mac, is it only necessary to look at the accuracy of the test set? Is the accuracy of the verification set needing to be concerned?

saichand07, 2019-03-21:
you can check the Test and validation accuracy as well, 
How did you create your dataset ??

ccnankai, 2019-03-22:
I used a laptop to record, and then cut into 1.5s audio file. Before quantification, the tensorflow program prediction is still very accurate. When quantifying, quant_test.py shows more than 90%, and it is impossible to predict when moving to the MCU.If there is a little sound, it will show wanted words.Have you run successfully on mcu?

ccnankai, 2019-03-22:
The accuracy   on the test set is very high, but the actual environmental effect is particularly poor, showing a word and rarely changes.

ccnankai, 2019-03-22:
I trained speech_commandsv2. Accuracy is 94% on the test set, but on stm32f7x, no keywords are detected.

saichand07, 2019-03-22:
@ccnankai 
Currently, I also facing the same problem,  Tensorflow models was running really well, but when it comes to the realtime test, my MCU is detecting some random words until and unless there is an input sound. 

I am using Analog microphones to give the input sound, I think problem is with my analog microphones
, I want to switch to Digital I2S microphones but I don't know how to integrate it with my S32K148 Eval Board.

if do you have any ideas, please share here

ccnankai, 2019-03-22:
@saichand07 
It's not a microphone problem. I extracted the code and ran it on the laptop. The generated weigths.h prediction result is wrong. However, the weights.h provided by this project is very accurate.Is the quantization process wrong, first fold_batchnorm.py then quant_test.py, right?

saichand07, 2019-03-22:
@ccnankai 
Currently, I am using the weights which were provided by this project only, But I also tested my microphone output with the Oscilloscope, the output amplitude is very poor
the quant process provided by them was right, 
yes first run the batch_normalization then run the quant_test.py
Did you run the simple test by giving a wav file array as input ?

ccnankai, 2019-03-22:
@saichand07 

I ran simple test  with my own generated weights.h and compared it with the .pb model output. The results were all wrong. The weights.h provided by this project is very accurate on stm32f7. I would like to ask if the weights.h you have trained yourself get  good result

saichand07, 2019-03-22:
I haven't tried real time test but the simple test got perfect results, 
you can check once again with your weights, follow the same format they have provided.

ccnankai, 2019-03-22:
@saichand07 
Ok, I will try it first. There should be an error in the intermediate steps of generating weights.h.

ccnankai, 2019-03-23:
@saichand07 
 I am sorry to bother you, how do you calculate the bias shift and output shift?

closed 2019-03-24 by ccnankai

saichand07, 2019-04-17:
@ccnankai 
did you find the reason, why model is detecting wrong keywords on Board?
what did you change ?